FreeCAD/FreeCAD #7305: [Bug] OpenSCAD: Import produces extra "Lines" & one extra "Layer"

gneiss15, 2022-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=10&t=70761

### Version

0.21 (Development)

### Full version info

```shell
FreeCAD_weekly-builds-29485-2022-07-24-conda-Linux-x86_64-py310.AppImage
```

### Subproject(s) affected?

OpenSCAD

### Issue description

As written in the title, the imported Object contains some extra parts (Lines & layer).
(There are other import-bugs, but I was requested to file a separate report for each)

I had attached two scad-files.
The one that exhibits the bug is the file "CR2032.scad".
The other one is imported in correct oriantation.

[Scad-Files.zip](https://github.com/FreeCAD/FreeCAD/files/9253638/Scad-Files.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-01:
Is there a screenshot of the bug available ? can it be pasted to this ticket please ?

chennes, 2022-12-02:
There are a few lines at the end of `importDXFface` that are commented out -- I'm not sure why. Those lines remove the extra draft lines that are being left in after importing text. A little bit of extra work needs to go into removing the final layer container.

chennes, 2022-12-02:
I believe this is fixed by b3817cc3a67e48a0571270403bf35053279a9332 -- please re-open if that is not the case.

closed 2022-12-02 by chennes