FreeCAD/FreeCAD #7317: [Feature Request] [TD]Calculate Area of Arbitrary Faces

WandererFan, 2022-08-05:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://www.forum.freecadweb.org/viewtopic.php?f=3&t=70824

### Subproject(s) affected?

Techdraw

### Idea description

ExtensionAreaAnnotation currently is limited to polygonal faces.  The ability to calculate areas for arbitrary faces is available in OCC.  ExtensionAreaAnnotation should be able to accept an arbitrary face (or list of faces).

It should also have an option to calculate the area of a bounding rectangle to handle situations where the selected faces contain voids.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-17:
Looks like this is where it is in the code: 

https://github.com/FreeCAD/FreeCAD/blob/f741d28b43b93a08ac2f3700e0df2e831b873576/src/Mod/TechDraw/Gui/CommandExtensionPack.cpp#L1812-L1820

WandererFan, 2023-11-17:
GProp_GProps gprops;
BRepGProp::SurfaceProperties(face, gprops);
double area = gprops.Mass();

luzpaz, 2023-11-17:
@pavltom if this is of interest to you, would you like to take a swing at it ?

commit 0e5aa8c4 (2023-11-24): [TechDraw] Fixes #7317 - Calculate Area of Arbitrary Faces
closed 2023-11-24 by WandererFan