FreeCAD/FreeCAD #7321: [Bug] <title>part_sweep and helix

Aleks-73, 2022-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

part_sweep and helix

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: Russian/Russia (ru_RU)
[/code]
```

### Subproject(s) affected?

Part

### Issue description

I tried to make helix : pitch 5mm, height 4,5mm, radius 20mm, the helix look as broken. When I make height more than 5mm it look normal.
[image]
[image]
[part_sweep and helix.zip](https://github.com/FreeCAD/FreeCAD/files/9275353/part_sweep.and.helix.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-08-06:
Thank you @Aleks-73 

We have a lot of 'sweep' related bugs, did you search them beforehand? Also our guidelines ask that users report issues to the forum.freecad.org first before opening a ticket, it saves the volunteers time when doing so.

grubertechnic, 2022-08-06:
related: [](https://forum.freecad.org/viewtopic.php?f=3&t=64333) and many more. The model is OK, the tessellation (representation) is not OK. Even ever cut works.

[image]
 
For that, there was never a ticket. I also thought with OCCT7.6.0 it vanished. But it seems, that there are stll some cases, where this issue occur.

Praveen-Rai, 2022-08-12:
[HelixFailure.zip](https://github.com/FreeCAD/FreeCAD/files/9329162/HelixFailure.zip)

I am facing a similar issue, while trying to create a square threaded bolt. [ Try entering a pitch of 17 mm in M6_SQ12_3Tr_12mm_Failing.fcstd ]

I am able to create one by creating a split sketch profile of the thread and later applying Polar Pattern. This file is also in the attached zip. The problem here is the Polar Pattern feature makes the part changes too slow, and often freezing is observed. For eg. try editing the dimensions in Sketch002 in M6_SQ12_3Tr_12mm.FCStd. Not sure if the combination of Helical Sweep and Pattern feature is the root cause, but as far as I recall, the Pattern feature did not present any such issue for me for other feature types ( Note : I haven't used Pattern feature extensively )

grubertechnic, 2022-08-13:
Please both make a thread in the forum.

luzpaz, 2022-08-23:
Please follow-up with forum thread or this ticket will be closed.

Praveen-Rai, 2022-08-25:
I've created a post on the forum. Here's the link [part sweep and helix presentation problems](https://forum.freecadweb.org/viewtopic.php?f=19&t=71382)

PLChris, 2022-08-25:
The same helix works in PartDesign
[image]
[PartDesign_sweep_cb.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/9421747/PartDesign_sweep_cb.FCStd.zip)

grubertechnic, 2023-07-29:
version

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33658 (Git)
Build type: Release
Branch: master
Hash: ba5f7e6d75dee09ec7076ad24895b04e56f21b63
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
```

The file from @Aleks-73  works flawless for me, but note, i am on >= OCCT7.7.0 . As far is i know, they improved the meshing of the tessellation. Need to verify.

[image]

-----------------------

The second file from @Praveen-Rai have nothing to do with the issue. The sketch of the hexagon is simply invalid. Delete the contend of the sketch and make a new hexagon from scratch and it works. Note, that the geometry check here also fails. Could be, that the thread shape is self intersection. Also there already some issues with helix already, but there also tickets for that already.

[image]

[image]

-----------------------
-----------------------

At least a candidate for closing, but verification needed.

closed 2023-08-25 by luzpaz