FreeCAD/FreeCAD #7322: [Bug] <title>Missing color info in Select Color dialogue.

Batucada2, 2022-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=615323#p615323

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * Assembly4 0.12.3
[/code]
```

### Issue description

When setting the colour via Appearance->Shape color in the Select Color dialogue, the previously assigned colors are not taken over into the input fields of the dialogue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-08-06:
@Batucada2 is this reproducible on the 0.21 development version ?

Batucada2, 2022-08-06:
I can't test because I'm only running 0.20
I'm pretty new to FreeCAD, how I can get 0.21?

luzpaz, 2022-08-06:
@Batucada2 New builds are generated weekly or even daily (via homebrew IIRC). See https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Hey @PLChris, can you reproduce on v0.21 on macOS ?

Batucada2, 2022-08-06:
I've just checked FreeCAD_weekly-builds-29865-2022-08-01-conda-macOS-x86_64-py310.dmg
There is no improvement regarding the problem noted here

Batucada2, 2022-12-22:
I have provided at least 2 posts with a very detailed description on the subject, but in German. I hope that the language will not be a problem.

See here:

https://forum.freecadweb.org/viewtopic.php?p=614996&sid=4d7e12321b592eb8cd33af335bc29c43#p614996

or

https://forum.freecadweb.org/viewtopic.php?p=615094&sid=4d7e12321b592eb8cd33af335bc29c43#p615094

I assume, however, that this is not a fundamental problem of FreeCAD, but rather to be found in the part of Qt.

Batucada2, 2023-02-13:
is there a new development on this topic? I don't want to push - but this already seems to be a very old unresolved issue.

luzpaz, 2023-02-13:
Why is this issue closed?

Batucada2, 2023-02-13:
Sorry. Maybe I made a mistake when I left my query? It was not my intention to close the case.

luzpaz, 2023-04-16:
@xtemp09 does #9287 reolve this ticket ?

xtemp09, 2023-04-17:
@luzpaz, yes.

closed 2023-04-17 by chennes