FreeCAD/FreeCAD #7327: [Bug] Program received signal SIGSEGV, Segmentation fault

ToddG, 2022-08-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

None

### Version

0.20 (Release)

### Full version info

```shell
$ ./freecad 
FreeCAD 0.20, Libs: 0.20R29177 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

User path has changed to /home/toddg/.local/share/FreeCAD/. Please move user modules and macros
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f2919271090]
#1  0x7f2814faa279 in TechDraw::ShapeExtractor::getShapes(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> >) from /tmp/.mount_freecazKAUpv/usr/lib/TechDraw.so+0x379
#2  0x7f2814ffcd34 in TechDraw::DrawViewPart::getSourceShape() const from /tmp/.mount_freecazKAUpv/usr/lib/TechDraw.so+0xe4
#3  0x7f2815006c02 in TechDraw::DrawViewPart::execute() from /tmp/.mount_freecazKAUpv/usr/lib/TechDraw.so+0xa2
#4  0x7f281501f159 in TechDraw::DrawProjGroupItem::execute() from /tmp/.mount_freecazKAUpv/usr/lib/TechDraw.so+0x129
#5  0x7f281501f193 in TechDraw::DrawProjGroupItem::onDocumentRestored() from /tmp/.mount_freecazKAUpv/usr/lib/TechDraw.so+0x13
#6  0x7f291a77b302 in App::Document::afterRestore(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADApp.so+0x642
#7  0x7f291a77c1ac in App::Document::afterRestore(bool) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADApp.so+0x4c
#8  0x7f291a9608e4 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADApp.so+0x1db4
#9  0x7f291a961138 in App::Application::openDocument(char const*, bool) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADApp.so+0xc8
#10  0x7f291a969fb8 in App::Application::sOpenDocument(_object*, _object*, _object*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADApp.so+0xe8
#11  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libpython3.9.so.1.0(+0x147085) [0x7f291ac11085]
#12  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x286) [0x7f291abffe76]
#13  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x5120) [0x7f291abfced0]
#14  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libpython3.9.so.1.0(+0x12d029) [0x7f291abf7029]
#15  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalCodeWithName+0x51) [0x7f291abf6ce1]
#16  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libpython3.9.so.1.0(PyEval_EvalCodeEx+0x3a) [0x7f291abf6c8a]
#17  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libpython3.9.so.1.0(PyEval_EvalCode+0x1c) [0x7f291ac8932c]
#18  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libpython3.9.so.1.0(+0x1eda9f) [0x7f291acb7a9f]
#19  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libpython3.9.so.1.0(+0x1e9adc) [0x7f291acb3adc]
#20  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libpython3.9.so.1.0(PyRun_StringFlags+0xa7) [0x7f291acab517]
#21  0x7f291a47cb8c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADBase.so+0x6c
#22  0x7f291b329b35 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADGui.so+0x95
#23  0x7f291b329ce1 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADGui.so+0x101
#24  0x7f291b24e080 in Gui::Application::open(char const*, char const*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADGui.so+0x4a0
#25  0x7f291b31c73d in Gui::RecentFilesAction::activateFile(int) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADGui.so+0x26d
#26  0x7f291b331351 in Gui::Command::_invoke(int, bool) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADGui.so+0x271
#27  0x7f291b3317e7 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADGui.so+0x127
#28  0x7f29197af39a in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Core.so.5+0x6c0
#29  0x7f2919fbe2b3 in QActionGroup::triggered(QAction*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5+0x2f
#30  0x7f29197af39a in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Core.so.5+0x6c0
#31  0x7f2919fbb958 in QAction::triggered(bool) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5+0x32
#32  0x7f2919fbd2e2 in QAction::activate(QAction::ActionEvent) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5+0xc6
#33  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5(+0x263689) [0x7f291a0af689]
#34  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5(+0x2681d8) [0x7f291a0b41d8]
#35  0x7f291a0b5a81 in QMenu::keyPressEvent(QKeyEvent*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5+0xb77
#36  0x7f2919feb59d in QWidget::event(QEvent*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5+0x1b3
#37  0x7f291a0b5fe2 in QMenu::event(QEvent*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5+0x2d0
#38  0x7f2919fc0822 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#39  0x7f2919fc4ceb in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5+0x467
[... 2 lines cut ...]
#42  /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5(+0x1aff9c) [0x7f2919ffbf9c]
#43  0x7f2919fc0822 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#44  0x7f2919fc634d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#45  0x7f291b2eac49 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADGui.so+0x89
#46  0x7f2919798f88 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Core.so.5+0x86
#47  0x7f2919b3aedc in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Gui.so.5+0xca
#48  0x7f2919b3d96b in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Gui.so.5+0x79
#49  0x7f2919b2ac95 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Gui.so.5+0x63
#50  /tmp/.mount_freecazKAUpv/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x59f09) [0x7f2914ba9f09]
#51  0x7f291979599b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Core.so.5+0x187
#52  0x7f2919799811 in QCoreApplication::exec() from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libQt5Core.so.5+0xfd
#53  0x7f291b26f75e in Gui::Application::runApplication() from /tmp/.mount_freecazKAUpv/usr/bin/../lib/libFreeCADGui.so+0x1b2e
#54  /tmp/.mount_freecazKAUpv/usr/bin/freecad(+0x67b1) [0x56545d1bb7b1]
#55  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f2919252083]
#56  /tmp/.mount_freecazKAUpv/usr/bin/freecad(+0x6af1) [0x56545d1bbaf1]
```

### Issue description

Simply opening this freecad project causes a seg fault.
[level.zip](https://github.com/FreeCAD/FreeCAD/files/9277223/level.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2022-08-07:
Body035 in group Drywell has no shape.  See https://www.forum.freecadweb.org/viewtopic.php?p=609773#p609773

This has been fixed by PR #7184 and will be included in v0.20.1.

Is there a reason you are using PartDesign Bodies as containers for Draft Clones?  App::Part seems more appropriate.

WandererFan, 2022-08-07:
[level-01_fixed.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/9277369/level-01_fixed.FCStd.zip)

remove the ".zip" for your file with Body035 removed.

luzpaz, 2022-08-10:
> This has been fixed by PR #7184 and will be included in v0.20.1.

Closing ticket.

closed 2022-08-10 by luzpaz