FreeCAD/FreeCAD #7334: [Feature Request] Dimension diameter with respect to symmetry axis

FEA-eng, 2022-08-08:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=70867

### Subproject(s) affected?

Sketcher

### Idea description

For revolved parts, it might be convenient to dimension the diameter instead of radius with respect to the symmetry axis.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2023-10-28:
The forum threads mentions a quality of life that would be useful: creating a line type for revolution axis, like Catia. `PartDesign_Revolution` or `PartDesign_Groove` could detect that one axis is present in the sketch and automatically present to the user the result around this instead of sketch axis and origin axis.

If implemented it would be difficult to find a color for the new line type. Anyway, there will be most likely an overhaul of the lines for legibility (see #10100) so with different thickness for construction and geometry, plus dashes for construction and dash-dots for axis the need for colors would be greatly reduced.

------

Here are some screenshots from Solidworks, extracted from the documentation, that illustrate the desired result to avoid the need to go back to the  forum:

[image]

[image]

As you can see in radius mode, there is no “R” prefix. I think that this prefix would be useful.

pierreporte, 2023-11-28:
@PaddleStroke what do you think of this?

I think that it could use the current diameter/radius tools, modified to accept other geometries than circles and arcs. Actually there should be possible do dimension the inner diameter of a torus for example, but starting with lines would be a great improvement already.

PaddleStroke, 2023-11-29:
This is a bit hard to implement and is not in my imediate priorities but I'll keep that in mind.

pierreporte, 2023-12-09:
This feature should work with either to-be-implemented axis and with the main axis of the sketch. These main axis could be defined with the new type, however it should then allow normal dimensioning, because you don’t want a diameter symbol when drawing a box.

FlachyJoe, 2023-12-13:
IMHO the axis-line is not needed and the constraint should work with construction segments.

pierreporte, 2023-12-13:
@FlachyJoe then `PartDesign_Revolution` should allow to pick up construction lines as the revolution axis, which means either displaying construction lines when outside a sketch (like Solidworks) or having a parameter to set for the desired line… which is equivalent to a new line type.

FlachyJoe, 2023-12-14:
@pierreporte Construction lines are already visible in the axis combobox
[image]

pierreporte, 2023-12-14:
@FlachyJoe Thank you for this discovery However this doesn’t really work when there is more than one construction lines. When the profile is complex and need construction geometry, it would be cumbersome to try everything. So I still think that there should be either all construction geometry displayed (like Solidworks) or a way to make a selected construction line visible from outside the sketch—which again is equivalent to a new line type.

maxwxyz, 2023-12-15:
maybe a separate command can create a *real* axis from the sketch and the axis could be constrained by current dimensions

FlachyJoe, 2023-12-15:
Other possibilities maybe easiest to implement:

- Add a sketch display mode which show construction lines and switch to this mode when axis is required. 
- Create temporary pick-able infinite line (Coin object) from construction lines.

pierreporte, 2023-12-15:
@maxwxyz this is something Alibre does (but not for the purpose we are discussing). It is interesting if the datum line moves when the sketch is modified.

@FlachyJoe I like this sketch display mode, and in fact it can be used to create datum lines because they can already be created along sketch lines. This special sketch mode could be activated automatically when performing an operation on the sketch (not only a revolution).

FlachyJoe, 2023-12-15:
@PaddleStroke @abdullahtahiriyo Do you think it's easily possible to put a sketch in a intermediate display mode same as edit mode but without user interaction (only selection allowed)?

abdullahtahiriyo, 2023-12-22:
@FlachyJoe 

Are you thinking into picking the line to be used as rotation axis from a sketch displayed on the screen while executing a PD operation, and thinking of using the sketcher to circumvent that the line segment defining the axis is of construction mode, and thus it is not shown outside of edit mode?

If yes, I think that is an overcomplicated approach.

RT's fork used to have a facility to "export" one segment from a sketch for external use. This functionality was "bundled" with his toponaming implementation and it did not make it into FreeCAD. Perhaps when the toponaming branch is added to FreeCAD, we can look into that (if wanted/necessary).

Regarding this issue (and forum thread), I see the following:
1. The Sketcher is a 2D tool often used to build 3D shapes, but not exclusively. Whatever we write in the sketcher needs to make sense in 2D. Half diameter quotes with full diameter value, IMO are very misleading. Full diameter quotes with a full diameter value I think it is ok.
2. We can enrich 2D geometry with metainformation without creating a new type (axis), by using a geometry extension (SketchGeometryExtension). This metainformation can be used at sketcher level to "paint labels and quotes" differently. At a 3D tool, it can be used for processing the sketch easier. For example, we could add metainformation to mark a line segment as an intended axis, and only show for selection in some 3D tools sketcher construction lines that have this metainformation marking. We could also add a name metainformation for better identification of specific axes.
3. A 3D tool could include lines with that metainformation in a combobox for selection, but could also represent in the 3D View, using coin, the lines having this metainformation (by properly transforming with the sketcher placement the geometry element).

FlachyJoe, 2023-12-22:
> Are you thinking into picking the line to be used as rotation axis from a sketch displayed on the screen while executing a PD operation, and thinking of using the sketcher to circumvent that the line segment defining the axis is of construction mode, and thus it is not shown outside of edit mode?

It's my thought and I see some advantage of showing dimensions and constructions when using a sketch in a 3D operation. This display mode could also be activated as highlight.

> 2. We can enrich 2D geometry with metainformation without creating a new type
> 3. A 3D tool could include lines with that metainformation in a combobox for selection, but could also represent in the 3D View, using coin

Great !

> 1. Half diameter quotes with full diameter value, IMO are very misleading. 

Maybe we can create a symmetric ghost geometry when an axis is declared. Should we then convert the sketch to a "rotational sketch" with specific behaviors?

Finally this discussion is splittable : this FR for a symmetry axis in sketch and the problem to pick the axis in the 3d view.

pierreporte, 2024-01-09:
> Half diameter quotes with full diameter value, IMO are very misleading.

This is valid in engineering drawings. The following image is from one of the two French bibles* about this and it explicitly advise about using those dimensions. The other bible and lesser used books do as well. They should be available in TechDraw as well if they aren’t already. They will be perfectly understood as long as the diameter symbol is present.

[image]

\* Jean-Louis Fanchon, *Guide des sciences et technologies industrielles*.