FreeCAD/FreeCAD #7338: [Bug]Cutting a sphere results in errors after merging with other objects

schoenid, 2022-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=70923

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: German/Switzerland (de_CH)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.56a
  * ArchTextures
  * Assembly4 0.12.3
  * BOLTSFC
  * CADExchanger
  * CurvedShapes 1.0.4
  * Curves
  * Defeaturing
  * Design456 0.0.1
  * dodo
  * EM
  * ExtMan
  * fasteners
  * FCGear
  * flamingo
  * kicadStepUpMod
  * MakerWorkbench
  * Manipulator 1.4.9
  * offline-documentation 1.0.0-alpha
  * ose-piping
  * OSE3dPrinter
  * parts_library
  * pcb
  * persistenttoolbars
  * Pyramids-and-Polyhedrons
  * pyrate
  * RemBench
  * retr3d
  * sheetmetal
  * Silk 1.0.0
  * slic3r-tools
  * ThreadProfile 1.81.0
  * workfeature
  * workfeature-macro
[/code]
```

### Subproject(s) affected?

Part

### Issue description

Cutting a sphere results in errors after merging with other objects.

Try this:

Torus:
R1: 10mm
R2: 5mm
Angle Y -90°
Position Z: -Radius1

Sphere:
R: <<Torus>>.Radius2
Position Z: <<Torus>>.Radius1

Box:
Length: <<Sphere>>.Radius * 2
Width: <<Sphere>>.Radius
Height: <<Torus>>.Radius1 + <<Torus>>.Radius2
Position X: -Length / 2

Cylinder001:
Radius: <<Torus>>.Radius1 + <<Torus>>.Radius2 + 1mm
Height: <<Torus>>.Radius1 + <<Torus>>.Radius2 + 1mm
Position Z: -<<Torus>>.Radius1
Angle X: -90°

Cylinder002:
Radius: <<Torus>>.Radius1 + <<Torus>>.Radius2 + 1mm
Height: <<Torus>>.Radius1 + <<Torus>>.Radius2 + 1mm
Position Z: -<<Torus>>.Radius1
Angle X: 135°

Fusion001:
Merge the Cylinders

Cut001:
Cut Torus with Fusion001.

The Result should be a small segment of 45° of the Torus.
[image]

Cut002:
Cut Sphere with Box
Position Z: -<<Torus>>.Radius1
Angle X: 45°

The Result should be a half a ball on the lower side of the Torus segment.
[image]

Fusion002:
Merge Cut001 and Cut002.

Result: Nothing - Resulting Shape is invalid, internal Name Fusion002.

Some different behaviour occurs, but also as unexpected results, when trying to merge Cut002 (the cutted sphere) with other objects, like cylinders, boxes and others. The sphere segment will never be displayed correctly, instead there is nothing, unexpected spikes and lines, symply the whole thorus, or other strange results.

Expected result:
[image]

Results:
Nothing
[image]

Merged Cut002 with Cylinder
[image]

Trying to merge this:
[image]
Result:
[image]
[image]
Components opaque and result semi-transparent:
[image]
Both figures should display the same!

This is obviously a bug.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

schoenid, 2022-08-08:
Workaround:

Calculate the posittion by using Trigonometry.
For 45° and symmetries sqrt(2) can be used ...
Use the U-Parameter of the sphere to cut it.
In many cases it may work. (... not all ...)
If still problems occur, rotations on other axes may solve it.

schoenid, 2022-08-08:
One strange thing:

Using an uncutted sphere with rotation 0° causes an error.
Using it with the opposite 180° works.
The same with 45° and 225° ...
Altough using an uncutted sphere leaves some strange artifacts and mirroring it, causes some strange  overhanging plane elements .?.!

I guess, that there are some trigonometric functions in use, which can cause "Division by 0" errors. This could be an explanation for such "side effects", if the "Division by 0" is not properly or separated handled, or at least the situations, where it may occur. (Just my opinion as software developer ... there are other possibilities, where the problems may come from. Unfortunately I don't have the time, to debug the whole code.)

luzpaz, 2022-11-20:
Closing ticket. Deferring to forum topic.

schoenid, 2022-11-20:
So, as I think, I gave enough information, to reproduce the problem.
The issue is obvious.
Closing the issue does not solve it.

FlachyJoe, 2022-11-21:
@schoenid The problematic file was asked here and in the forum.

FYI, sphere - box + torus works as expected here:
```
OS: Ubuntu 22.04.1 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.31028 (Git)
Build type: Unknown
Branch: master
Hash: 9d98347a51df63c297355d649a8cd9554737dc91
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: French/France (fr_FR)
```
[image]

adrianinsaval, 2022-11-23:
> So, as I think, I gave enough information, to reproduce the problem. The issue is obvious. Closing the issue does not solve it.

It would have been 100 times easier if you just shared the damned file, as it is one has to carefully create each object and follow your syntax to try and get the conditions, why do you choose to waste everyone's time??

Here a file with what I think this guy meant (remove dummy zip extension):
[7338.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10077468/7338.FCStd.zip)
attempting to make a union of the cuts fails for me. btw this is an unnecesarily complex way of making this shape

schoenid, 2022-11-23:
I've recreated the file.
[Sphere-Problem.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10079605/Sphere-Problem.FCStd.zip)
(Remove the dummy zip extension)

In the meantime I've learned, that segments of spheres and toruses can be done by the Angle3 attribute, so things do go easier. Also the problems with merging did not occur. (altough these are not soved and may occur in different, unexpected situations ...) But still in some cases I've got problems with merging, depending of the rotation of bodies with circular shapes. Often it helps then to turn it about 0.001 degrees.

With Part Design I've got mostly troubles, not to be able to fix missing points of freedom to complete sketches. 
That's the reason, why I'm working mainly in the Part workbench.

adrianinsaval, 2022-11-23:
I think this is just one more example of OCCT's troubles with coplanar boolean ops and seam edges. I don't think there's anything we can do on the FreeCAD side.

adrianinsaval, 2022-11-23:
[Sphere-Problem.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10079838/Sphere-Problem.FCStd.zip)
interestingly it works for me after recomputing everything

> 
> OS: Linux 6.0.9-zen1-1-zen (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.21.31081 (Git)
> Build type: None
> Branch: master
> Hash: 69dd6ea965ae0efd026855b8ba99e4c1465d96f5
> Python 3.10.8, Qt 5.15.7, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
> Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
>

luzpaz, 2023-02-26:
What's the status of this ?

schoenid, 2023-03-04:
I still have massive problems with roundings, like the following:
[Test.zip](https://github.com/FreeCAD/FreeCAD/files/10889300/Test.zip)
It seems to be impossible, to create this object in parts workbench.
[image]
I found no way, to merge the objects into one.

FlachyJoe, 2023-03-04:
@schoenid Your cloned scaled cylinder brake the model.
[image]

NB : you should try PartDesign and sketch based modelling
[image]

schoenid, 2023-03-04:
@FlachyJoe No, not by the scaled cylinder, but by the fillets!
Anyway, thank you for the attempt to help!

I've tried to recreate the whole parts on different ways and found again a workaround, not the best solution, but usable.
[Test2.zip](https://github.com/FreeCAD/FreeCAD/files/10889600/Test2.zip)
The Problem occurs, if the fillet radius of the center part is equal or smaller than the fillet radius of the side parts.
[image]
Why it occurs, if the radius is the same, makes no sense to me. But if I set it equal (Fillet BlendeEPre  Kante11 _ 2.49mm) the side parts do disappear ... 
[image]
So the problem is always with fillets and/or radius. (On some projects even with sketches ...)

FlachyJoe, 2023-03-05:
> @FlachyJoe No, not by the scaled cylinder, but by the fillets! 

Clone002 generate an invalid solid (see my screenshot of [Part_CheckGeometry](https://wiki.freecad.org/Part_CheckGeometry) result) so Fillet is built on sand.

> I've tried to recreate the whole parts on different ways

Your workflow is definitely bad, with so many operations you multiply the error possibilities.
Your model graph:
[image]
Mine:
[image]

schoenid, 2023-03-05:
@FlachyJoe Really interresting!

it's my impression, designs created by parts workbench will create always invalid solids with time ...
But unfortunately I have always problems, creating sketches. Often there is no solution to solve all the free things of freedom (I always forget the expression), so it can't be completed. And Other problems. For me the workflow does not correlate with my own thinking, causing troubles and lots of time to solve it. Far to much time.

The workflow in the parts workbench is more, like I think, by taking base objects and create something out of it by additive or subtractive operations in space. This is easy for me and takes not much time. And as long, as I don't need dimensioning ... otherwise I take Visio and draw it there ...

Another thing in your graphs, showing the complexity, is the depency between objects, as the size of objects depends from the dimensions of others. Basically the boxes Blende* gives the width and height of the whole thing, the boxes Träger* gives the length, the boxes Verbinder*  and its values in parameters can completely be removed. So the complexity of my model decreases and yours increases.

My workflow comes from the possibilities in the parts workbench. Sometimes operations can't be done on fusions, fillets and others, then first a clone is required. In larger projects I often create basic forms as separate subprojects, then I use just only a simple copy in the main project to keep the amount of data small. I'm always learning and things are getting better and better. 

But thank you for your view to the complexity of the model and the way, you are working. This gives me motivation, to learn more about the partdesign workbench.

adrianinsaval, 2023-03-08:
this is derailing into a conversation that should be had in the help forum rather than an issue tracker...

Besides that, the newly given examples aren't really comparable, the files from the OP work after a full recompute on current versions so this should be closed.

schoenid, 2023-03-11:
Short summery:

As some radii are involved in calculations while cutting, fillets or other operations, the process fails often and leads to errors in par workbench. The problem still persists and no solution is in view.

schoenid, 2023-03-11:
There seems to be no intention to solve this problem, so I'm looking to try Fusion 360.

closed 2023-03-11 by schoenid

adrianinsaval, 2023-03-11:
as with the original file the second file also works in current versions after a full recompute
[Test_fixed.zip](https://github.com/FreeCAD/FreeCAD/files/10948922/Test_fixed.zip)
```
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32220 (Git)
Build type: None
Branch: master
Hash: 088f51632b37279b74be1cfcc1a94779734a53bd
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
```

however it is clear that this is not a good workflow:
[image]
It might have been better to do the fillets as last step here (but taking into account that fillets are known to be sensitive, this is a known upstream issue that cannot be solved at our end)

> But unfortunately I have always problems, creating sketches. Often there is no solution to solve all the free things of freedom (I always forget the expression), so it can't be completed. 

this is a common misconception but you don't really need to constraint all degrees of freedom in a sketch to use it (although it is desirable).

> There seems to be no intention to solve this problem

there's nothing to solve, the files are already working despite being built suboptimally.

schoenid, 2023-03-15:
Working in part workbench is a more natural way, to get results.
If you want a hole in a bowl, you drill it with a cylindrical tool.
You will not gow and construct the bowl by transform it out of a circle.
Then to add fillets, you will take a piece of sandpaper an wipe out the sharp edges.
You will not go and think about, how to construct it in to a base form and then expand it to 3D ...
So to use parts with boolean operations is normally a morr natural way and much, much faster.

... if the CAD tool works as expected - > not like FreeCAD.
I've tried Blender now and it's a lot more user friendly for a natural workflow, altough some other problems come with it ...

About the picture 
[image]
It's clearly not the workflow, that produces such artefacts. It's the CAD system !!!!!