FreeCAD/FreeCAD #7341: [Bug] <Sketcher  Auto-Constraints  not  functional  and  or  intermittent>

CaillouxFred, 2022-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=23201&start=10

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: French/Canada (fr_CA)
```

### Issue description

When setting the Auto Constraints in Edit/Preferences/Sketcher,
Tick ON Auto Constraint will not necessarily be activated when attempting to create a Polyline closed form.
I’ve included a sketch for repeating the problem.
The web shows many users, since many years ago, experiencing the same problem.
Even a YouTube video addresses the problem:  “FreeCAD Autoconstraint / Snap Problem”

The proposed sketch involve the attempt to make a Bolt.
When entering the “sktThread” sketch one can see that the “Edit controls”  Auto constraints is selected ON.
The problem manifest ( but not necessarily limited to ) when a closed Polyline is created.
In this example try to create a triangle utilizing the Polyline tool.
Because the background plane is XZ with X pointing to the right, by default, the attempt can be successful.
If the construction view is rotated on the Z axis such that X is pointing to the left (180°), then the Polyline cannot be closed.
Note: The XZ plane is rotated utilizing the upper right tool on the sketcher window.
Trying to build a triangle utilizing the Polyline tool will result in a semi triangle for which the End of the third segment is impossible to connect with the Start of the first segment, as if the Auto-Constraints feature has suddenly been disabled. 
In the previous case, where the Z axis has not been twisted 180°, there is no problem closing the triangle, where in this case the triangle is impossible to close.
If the XZ plane is rotated back to X axis pointing right, then the Polyline tool can perform correctly and closing the triangle suddenly become possible again.
Many experiments have been tried to see if there could be interactions with either, the Auto Constraints ticked ON from the preferences or within the Edit Controls from the Sketch window dialog. Both options are erratic when the Z axis is manipulated. 
It would appear that the only way for the Auto Constraints feature to be functional is when no modification of the Z axis rotation has been done.
In a perfect world, it is obvious that Auto-Constraints should have nothing to do with the fact that a plane is or not rotated.

### Anything else?

“FreeCAD Autoconstraint / Snap Problem”     [https://youtu.be/suFp5-SWSA4](url)

Here is my Test File : [_AutoConstraints.zip](https://github.com/FreeCAD/FreeCAD/files/9284379/_AutoConstraints.zip)
[image]
[image]
 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

CaillouxFred, 2022-08-08:
I have found this FreeCAD subject discussed on the forum but not as concise and no proper solutions were proposed. At least, I have not found any. Despite some lengthy discussions it would appear that no other Bug Report address the subject precisely. Hope that this will lead to an even better FreeCAD.  Thanks to all for the great work.

CaillouxFred, 2022-08-09:
Sorry for not reading about the policy to post on Forum prior to report a bug.
I just added the document to the Forum now :https://forum.freecadweb.org/viewtopic.php?f=3&t=70950
Please put on old for now.
Perhaps someone will come up with a solution.

CaillouxFred, 2022-08-17:
[quote=FredCailloux post_id=618156 time=1660599945 user_id=43358]
[quote=chrisb post_id=617622 time=1660433933 user_id=5646]
I think the issue should be something like this:
[i]
Description of the problem: The points in Sketcher are raised above the geometry in order to make the selectable with the mouse. This works well in the default orientation, but fails if the sketch is turned so that it is seen from the back.

Feature request: Make the points selectable with the mouse independent from the orientation of the sketch. This could e.g. be done by raising them always towards the user.
[/i]
[/quote]
Sounds good to me
[/quote]

adrianinsaval, 2022-08-17:
There is already an issue for the underlying problem: #5673, at most the info that it prevents auto constraints from working should be added there.

luzpaz, 2022-08-17:
@adrianinsaval shall i close this ticket in favor of that one?

adrianinsaval, 2022-08-17:
IMO yes

luzpaz, 2022-08-23:
assigned to Abdullah, hopefully he is well and has some free time to weigh-in on this and other Sketcher related issues soon. :crossed_fingers:

adrianinsaval, 2022-09-09:
fixed by #7417 this can be closed

luzpaz, 2022-09-10:
Fixed in 968fb5b  
Thanks @adrianinsaval  
h/t to @0penBrain for the fix

closed 2022-09-10 by luzpaz