FreeCAD/FreeCAD #7342: [Bug] shape binder selection color persists after task close

quiret, 2022-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=70663

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/Germany (en_DE)
```

### Issue description

**Problem description:**
The selection color of shape binders stays on the sketcher geometry even though the task which colored it no longer exists. After deleting the shape binder the color persists aswell. This problem even persists across saving the FreeCAD document.

[https://youtu.be/PKffpxEcb-A](YouTube)

**Steps to reproduce:**
1. create a sketcher geometry
2. create a shape binder which selects the previously created sketcher geom,
on a edge-by-edge basis
3. click on Add Geometry and then click OK. You will see that the geometry
is purple even though we have left the task
4. edit the shape binder and remove one edge, then click OK. You will see
that the geometry is no longer purple.
5. edit the shape binder, then click Add Geometry and click OK. The geom
is purple again, minus the one edge.
6. again edit the shape binder, then click Add Geometry, click on the previously
removed edge, adding it back into the list, but click Cancel instead. You
will see that the geometry is purple.
7. again edit the shape binder, then click on Add Geometry, finally click on the
previously removed edge. This time the geometry stays purple even though
you have added the last one to the list.
8. click on OK, then delete the shape binder from the FreeCAD document. You
will see that the sketcher geometry stays purple even though the shape
binder which created the selection color no longer exists.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-06:
@quiret the youtube clip isn't available anymore. Can you upload the clip directly into github ?

luzpaz, 2023-11-17:
@FEA-eng can you re-confirm this ?

FEA-eng, 2023-11-18:
> @FEA-eng can you re-confirm this ?

Yes, I can confirm that it still happens.

[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35046 (Git)
Build type: Release
Branch: main
Hash: 6a66034ce27290e7b385a933d77375f55492d194
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

FEA-eng, 2024-12-14:
Still relevant

luzpaz, 2024-12-17:
@quiret any chance you have the video clip around and can upload it to the this ticket? (the YT link is broken)

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/a8e30010a8f9348c8a8c88746498bcba3c54dae2

ShapeBinder restores the highlighted source appearance when editing ends, even when Support was changed or cleared.