FreeCAD/FreeCAD #7363: Help view en  FR

FMD14, 2022-08-13:


luzpaz, 2022-08-13:
Transferred back to main FreeCAD repo as this is a bug somewhere. Perhaps in the external help addon.

FMD14, 2022-09-06:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30256 (Git)
Build type: Release
Branch: master
Hash: 55866fd62221beadad3cfd60915278b156609a66
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: French/France (fr_FR)
[/code]

Hello
With the latest version 0.21 on a computer works without doing anything
On the second I want to install at the time install the help I have an error 
and it does not install the additional module

[image]

luzpaz, 2022-09-06:
Would you be willing to test this on the v0.21.x dev branch ?

luzpaz, 2022-09-06:
@chennes have you happen to seen an addon manager error like this before :point_up:

FMD14, 2022-09-06:
I can test on the v 0.21x development branch
how to do it

luzpaz, 2022-09-06:
Available from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

FMD14, 2022-09-07:
Hello
Same error with downloaded version
Last night
Installation help the additional module does not install blocking

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30265 (Git)
Build type: Release
Branch: master
Hash: c35244e74f326a04c5bcab31e984742db582ac78
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: French/France (fr_FR)

[image]

luzpaz, 2022-09-07:
@yorikvanhavre have you seen this type of issue with the external help addon before?

luzpaz, 2022-09-09:
@yorikvanhavre please transfer to https://github.com/FreeCAD/FreeCAD-Help/

chennes, 2022-09-10:
The Addon Manager problem is on me -- when refactoring I missed a couple variable initializations. Fixed in b97c93008e.

luzpaz, 2022-09-10:
@FMD14 is there a way you can test this fix ?

FMD14, 2022-09-12:
What should I do?

luzpaz, 2022-09-12:
@FMD14 this should be available in the [weekly build](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds), if you want to test

FMD14, 2022-09-13:
error version 11/09/2022
[image]

luzpaz, 2022-09-13:
Please Always always add full [About](https://wiki.freecad.org/About) info. Not sure if you updated the actual updated weekly, your About info would tell us that instead of creating more confusion.

FMD14, 2022-09-13:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30345 (Git)
Build type: Release
Branch: master
Hash: f42a47aa0fd8c4f34ef63c185d9f29b9b6e84737
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)

yorikvanhavre, 2022-09-16:
Sorry about the delay guys! Indeed this is a help addon issue (I concentrated on the english side so far and didn't test other languages much yet). I'm transferring the issue there and do a fix ASAP