FreeCAD/FreeCAD #7370: [Bug] Arch Wall not visible when base sketch lines overlap

eshattow, 2022-08-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=23&t=71107

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux bookworm/sid (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.30058 (Git) AppImage
Build type: Release
Branch: master
Hash: cb96332a64db8ae855ca700212eebb0f5151230a
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
  * parts_library
[/code]
```

### Subproject(s) affected?

Arch

### Issue description

Create sketch, apply Wall. Edit sketch and overlap a new line within one of the existing lines. In the view your wall is not visible anymore. This seems broken, overlapping lines should get merged not cause a failure of the wall to be visible.

### Anything else?

[image]
[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2022-09-05 by yorikvanhavre

donovaly, 2022-09-12:
Is the bug also in FreeCAD 0.20? @Roy-043

Roy-043, 2022-09-12:
See comment here: #7436

PR #7436 (2022-09-05): Arch: Fix wall baseface generation for overlapping and self-intersecting traces
To fix the issue a new argument, `per_segment`, was introduced for the `bind` function in `faces.py`. Additionally the call to `face.fix()` was moved to that function.

Fixes #7370

Forum topics:
https://forum.freecadweb.org/viewtopic.php?f=23&t=67107&start=30
https://forum.freecadweb.org/viewtopic.php?f=23&t=71107

- [ ]  Your pull request is confined strictly to a single module. That is, all the files changed by your pull request are either in `App`, `Base`, `Gui` or one of the `Mod` subfolders. If you need to ma [... cut]