FreeCAD/FreeCAD #7373: navigation cube not navigate to 3d view

sedatkrn, 2022-08-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=71167

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30089 (Git)
Build type: Release
Branch: master
Hash: 335196a468fcf069f365dcbb24279a8295cc4683
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: Turkish/Turkey (tr_TR)
Installed mods: 
  * A2plus 0.4.56a
  * Assembly3 0.11.3
  * Assembly4 0.12.3
  * Behave-Dark-Colors 0.1.1
  * BIM 2021.12.0
  * CurvedShapes 1.0.4
  * Curves 0.4.2
  * dodo
  * fasteners 0.4.2
  * fcgear 1.0.0
  * sheetmetal 0.2.56
[/code]

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: Turkish/Turkey (tr_TR)
Installed mods: 
  * A2plus 0.4.56a
  * Assembly3 0.11.3
  * Assembly4 0.12.3
  * Behave-Dark-Colors 0.1.1
  * BIM 2021.12.0
  * CurvedShapes 1.0.4
  * Curves 0.4.2
  * dodo
  * fasteners 0.4.2
  * fcgear 1.0.0
  * sheetmetal 0.2.56
[/code]

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.709.28846 (Git)
Build type: Release
Branch: LinkStage3
Hash: c880c050c5f7601e4bae6740de71500edf2be0c7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Turkish/Turkey (tr_TR)
```

### Subproject(s) affected?

Core

### Issue description

The navigation cube is completely insensitive. It does not work with clicks, it can only be changed.

Open FreeCad
Start a new file
To change the view, click on the perspectives in the cube
There are no results

[image]

### Anything else?

i Although I deleted and restart the configuration files, there was no change.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-08-17:
Maybe it's a Windows issues? Can't recreate on:
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.30118 (Git) Snap 216
Build type: Release
Branch: master
Hash: ee68391bc8f37d7cc2ef74e0b674b27153cb73c4
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 1.0.4
```

adrianinsaval, 2022-08-17:
Description is too vague, can you show a gif what is happening? It works without issues for me on windows. Also, you are supposed to discuss this FIRST on the forum, then if we can confirm that it is a bug in FreeCAD you can create a github issue.

Roy-043, 2022-08-17:
I can't reproduce either.

@sedatkrn 
Try removing all addons to check if any of them are causing the issue.

You may also want to try deleting this Windows Registry key:
`HKEY_CURRENT_USER\Software\FreeCAD`
A number of Qt related settings are stored there.

sedatkrn, 2022-08-18:
> 

i did all but same :(

sedatkrn, 2022-08-18:
[image]

luzpaz, 2022-08-18:
@donovaly do you mind trying to reproduce ?

sedatkrn, 2022-08-18:
I did not encounter control and any problems on the other computer. I think there is a problem with my computer

closed 2022-08-18 by luzpaz

Alexdad76, 2022-09-24:
This solution helped me:
https://forum.freecadweb.org/viewtopic.php?p=414094&sid=26c0ebfbccecd45b4a9c8e8c34c97e2c#p414094

luzpaz, 2022-09-26:
Display driver bug. Out of scope of FreeCAD.