FreeCAD/FreeCAD #7382: [Bug] Can't report a bug, because I can't start a forum discussion. Why do I even need to start a forum discussion for technical bug!!!

dr3adl0rd, 2022-08-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

Not able to start one!

### Version

0.20 (Release)

### Full version info

```shell
Whatever phpBB you're using.
```

### Issue description

Your approach to asking people for forum discussion to every single type of FC bug is a serious obstacle, but ok, I tried to register with my e-mail domain I use for nearly 15 years, which I know works as I get 20 e-mails a day, and I pay for it, but your bloody "obstacle" system complains that it has no MX record. So, there it is, your bug reporting approach is flawed (buggy)!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-08-20:
Hello, you seem frustrated. Let me explain why we have this guideline in place:
1) historically freecad has had very long release cycles with very few backport minor releases. Many bugs get fixed in the dev branch and may not get backported to stable. Therefore bugs may be already fixed in dev. It creates a bug triage nightmare to handle redundant bug reports. 
2) historically, the forum is the fastest way to get bugs confirmed and fixed. Many bugs will be fixed on the spot.
3) The forum is a spam target, first time postings are automatically moderated and pending until OK'd by a moderator. After that, the poster can post freely.  

If you want to report bugs, consider also running the development version of FreeCAD (at this time it's v0.21.x That way you can confirm that said bug is not present in the bleeding edge version. Of course you need to make sure you backup any important projects before you open them in development because they may not be backward compatible (when you open them in an inferior version). 

We are all unpaid volunteers. We have things set up this way to save time for triagers, testers, and devs. If you still are not satisfied, perhaps you can step-up and become more active in the community to be able propose and change the current workflow. If not, that is understandable as well.

closed 2022-08-20 by luzpaz

dr3adl0rd, 2022-08-20:
So as you can see, I didn't mind the forum part so much, but the issue is still there, I can't register, and Contact us doesn't work either. But I see that nobody cares, so I think it's time to give up.

luzpaz, 2022-08-20:
@dr3adl0rd i checked to see and it seems like your post isn't in moderation. You're saying that the post is being rejected ?

In this case, please open another ticket with the bug in question.

dr3adl0rd, 2022-08-20:
It's the registration to forums that doesn't work. It tells me that the email address domain doesn't have MX record, which is wrong, as it does, and I even use it as we speak to get notifications of sub to this issue topic.
I also tried the Contact us on the same forums page (bottom of the page) and it says that another e-mail can't be sent at this time.

luzpaz, 2022-08-20:
Sorry, phpbb is a PITA. You can contact me at luzpaz at protonmail dot com with your info and I can look in to it. In the meantime, please open a new ticket with the bug in question. TIA