FreeCAD/FreeCAD #7386: [Bug] Unable to enter third decimal for scale

mpetrasinovic, 2022-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?style=3&f=35&t=58195

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.56a
  * ExternalAPI
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

It is not possible to enter the third decimal for scale property using GUI, which is too constraining. Both Techdraw and Draft Scale are affected.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2022-08-22:
This is a characteristic of the property editor.  It can't be fixed in TD or Draft.  The property editor uses the # of digits setting from the Units tab in Preferences > General. 

I would say this is not a bug, but a feature request for a new property type whose editor doesn't use the system setting for # of digits.  Only after such a property exists could we change TD and Draft to use it.

mpetrasinovic, 2022-08-24:
The original author on the forum suggested switching from PropertyFloatConstraint to PropertyPrecision, and I think it makes sense. I would not say this is a feature request, as the scale parameter is implemented as a feature, but its input field makes it unusable in some cases.

luzpaz, 2023-11-18:
bumping for discussion

Syres916, 2023-11-19:
See https://forum.freecad.org/viewtopic.php?t=82762 there's more work being done than was actually requested by @mpetrasinovic the decimals are set at 6 (purely from real world testing of a large assembly) and if the code detects a non-integer scale then it automatically creates an expression made of the fraction such as `37 / 36` in the Projection Group, Section View and Complex Section View.

There may well be other places in the code that need modifying but let's get the proof of concept accepted and take it from there.

Syres916, 2024-01-21:
@luzpaz / @Roy-043 this can be closed IMHO on the basis that the Section and ComplexSection now have 6 decimal places as a result of https://github.com/FreeCAD/FreeCAD/commit/b450a62ef5a1a06ffc65bf15858741c2d5d60bda

PR #12089 (2024-01-25): Draft: Draft_Scale: make sure scale inputboxes show at least 6 decimals
Fixes #7386.
commit 520b8bab (2024-01-25): Draft: Draft_Scale: make sure scale inputboxes show at least 6 decimals

Fixes #7386.
closed 2024-01-25 by Roy-043