FreeCAD/FreeCAD #7387: [Bug] Part: FeatureExtrusion.cpp issue with Draft_Wires

Roy-043, 2022-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=619535#p619535

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.21.30143 (Git)
Build type: Release
Branch: master
Hash: a72bf218f9917df37e811d7e73a36333282028f5
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * BIM 2021.12.0
  * CubeMenu
  * Curves 0.4.4
  * fasteners 0.3.49
  * fcgear 1.0.0
  * Help 1.0.3
  * Importer3D
  * InventorLoader 1.3.0
  * parts_library
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.55
  * TemplatePyMod
```

### Subproject(s) affected?

Part

### Issue description

The `Extrusion::calculateShapeNormal` function assumes that the shape of a [Part_Part2DObject](https://wiki.freecadweb.org/Part_Part2DObject) always lies on the XY plane of the coordinate system defined by its Placement. For a [Draft_Wire](https://wiki.freecadweb.org/Draft_Wire) that need not be the case.

Test code:
```
import FreeCAD as App
import Draft

doc = App.ActiveDocument

pts = [App.Vector (0.0, 0.0, 0.0), 
       App.Vector (5.0, 0.0, 3.0),
       App.Vector (5.0, 5.0, 0.0)]
       
wire1 = Draft.make_wire(pts)
wire1.Closed = True
wire1.MakeFace = False
doc.recompute()

extr1 = doc.addObject("Part::Extrusion", "Extrude_wire1")
extr1.Base = wire1
extr1.DirMode = "Normal"
extr1.LengthFwd = 2.0
extr1.Solid = True
doc.recompute()
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-05-08:
Draft Wires are not necessarily 2D (I use this often, for ex. to draw plumbing lines). IIRC they are Part::Feature, not Part::Part2DObject. Even if the automatic shape direction does not work, Extrusions still work ok...

Not sure how to proceed here... But it's not rally a bug IMHO

yorikvanhavre, 2023-05-08:
I'll remove this one from the 0.21 release project because I think it's deeper than just a bug fix

Roy-043, 2024-12-14:
This is still an issue.

Roy-043, 2025-05-28:
From the point of view of the Draft WB the ideal solution would be to replace:
`isDerivedFrom(Part::Part2DObject::getClassTypeId())`
with:
`isDerivedFrom(Sketcher::SketchObject::getClassTypeId())`

But (I think) that would mean having to add this:
`#include <Mod/Sketcher/App/SketchObject.h>`
And creating a dependency for the Part WB that is probably not allowed.

Removing the code block entirely would make the code less efficient for the majority of use cases.

@yorikvanhavre:
I think we need to fix this in the Draft WB. Using `Part::Part2DObject` as the base for objects that can be 3D is not a good choice. If I am not mistaken we need to use `Part::FeaturePython` with an attachment extension instead. Modifying this may prove to be tricky. Certain code may break. And converting old objects may be problematic (you may have to recreate them at the cost of breaking references).

PR #21636 (2025-06-02): Draft/BIM: change BezCurve, BSpline and Wire to Part::FeaturePython
Fixes: #7387.

See comment: https://github.com/FreeCAD/FreeCAD/issues/7387#issuecomment-2915599566

PR to change the base object of BezCurve, BSpline and Wire to `Part::FeaturePython`. This will only affect new objects. So code will have to also keep handling the old object type (`Part::Part2DObjectPython`).

The modification of BimPreflight.py needs to be verified. The steps in the old code lacked logic IMO. But I may have misunderstood.
commit e4adfc63 (2025-06-02): Draft/BIM: change BezCurve, BSpline and Wire to Part::FeaturePython

Fixes: #7387.

See comment: https://github.com/FreeCAD/FreeCAD/issues/7387#issuecomment-2915599566

PR to change the base object of BezCurve, BSpline and Wire to `Part::FeaturePython`. This will only affect new objects. So code will have to also keep handling the old object type (`Part::Part2DObjectPython`).

The modification of BimPreflight.py needs to be verified. The steps in the old code lacked logic IMO. But I may have misunderstood.
closed 2025-06-02 by yorikvanhavre