FreeCAD/FreeCAD #7390: [Bug] <Expression values not updating>

JensDyvik, 2022-08-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=71309&sid=5f08a91bbae06901242226315974586b

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: Norwegian Bokmal/Norway (nb_NO)
Installed mods: 
  * A2plus 0.4.54b
  * Assembly4 0.12.0
  * Curves
  * Defeaturing
  * DynamicData 2.46.0
  * ExplodedAssembly
  * fasteners 0.3.38
  * FCGear
  * ImportNURBS 1.0.0
  * kicadStepUpMod 10.13.0
  * Launcher
  * LCInterlocking
  * Manipulator 1.4.5
  * parts_library
  * Silk 1.0.0
```

### Issue description

I have trouble with the expression editor. I some cases the value does not update when the expression, or referenced values in the expression, are changed. Here is a screenshot showing how the computed result of the expression does not propagate to the constraint value:
[image]

This is the location of the troublesome Sketch in the file:
[image]

Here is the FreeCAD file file: https://github.com/Fabricatable-Machines/Humphrey2/blob/main/humphrey2.FCStd

The file uses a Link to this FreeCAD file, it might be needed to properly open the file:  https://github.com/Fabricatable-Machines/Humphrey2/blob/main/cnc-friendy-rack-and-pinion-for-humphrey2.FCStd

### Anything else?

- Giving the constraint a name fixes the problem. If the constraint name is removed the problem persists
- Opening the same file in FreeCAD version 19.4 (WIN10) and editing the constraint fixes the problem (no need to give constraint name)
- The problem persists when opening the file in FreeCAD version 20.1 (WIN10)
- I have only had this problem with this file. When I make a new project I am not able to replicate the bug/problem so far
- I also have the problem with some expressions driving the placement of a Body in this file
- I recently started using dual screens with separate display scaling settings
- The problem seems similar to this older developer forum post. But in my case clearing the expression does not fix the problem: https://forum.freecadweb.org/viewtopic.php?f=10&t=25707

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PLChris, 2022-11-26:
The bug occurs also outside of Sketcher, see https://forum.freecadweb.org/viewtopic.php?f=8&t=73898 and the file attached to the first post

xtemp09, 2023-01-31:
I can confirm that this bug exists on my platform. This problem is severe.

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

xtemp09, 2023-01-31:
The bug is connected to this part of a log:
```
Log: Sketcher::setUpSketch()-T:0.001
Log: Sketcher::Solve()-DogLeg-T:0
Log: Sketcher::setUpSketch()-T:0.001
Log: Sketcher::Solve()-DogLeg-T:0
```

xtemp09, 2023-02-03:
I can confirm that this bug is fixed in the today's weekly build. Perhaps, when the stable version is released, the issue should be closed.
```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31739 (Git)
Build type: Release
Branch: master
Hash: 37dc6e8a296073bf5c437b975d8fb236f923723f
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

chennes, 2023-02-04:
@JensDyvik can you confirm that this week's build fixes the issue for you as well?

JensDyvik, 2023-04-11:
@chennes I am sorry but I am not able to confirm. I have tried multiple nightly builds on Win10 x64 and I could not make any of them launch unfortunately.

adrianinsaval, 2023-05-04:
@JensDyvik did you maybe have the splashcreen disabled? there was a bug wit that on windows, that is fixed now in the code but I'm not sure if there is a build available with that fix, can you try again with the latest weekly build and if it doesn't launch try enabling the splashcreen from a working version or clearing your preferences

adrianinsaval, 2023-05-08:
I cannot reproduce this anymore either so I'll close this as fixed, you may reopen if you are able to reproduce the problem on a recent development release

closed 2023-05-08 by adrianinsaval

JensDyvik, 2023-10-26:
I can confirm the bug is fixed in release 21.1 on my system.

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Norwegian Bokmal/Norway (nb_NO)
Installed mods: 
  * A2plus 0.4.54b
  * Assembly4 0.50.4
  * Curves
  * Defeaturing
  * DynamicData 2.53.0
  * ExplodedAssembly
  * fasteners 0.3.38
  * FCGear
  * ImportNURBS 1.0.0
  * kicadStepUpMod 10.20.7
  * Launcher
  * LCInterlocking
  * Manipulator 1.4.5
  * Rocket 3.2.3
  * sheetmetal 0.3.1
  * Silk 1.0.0
[/code]

oldschoolcowboy, 2026-08-19:
OS: KDE Flatpak runtime (Wayfire/wayland)
Architecture: x86_64
Version: 1.1.3.44987 (Git)
Build date: 2026/07/25 04:52:02
Build type: Release
Branch: (HEAD detached at 1.1.3)
Hash: 145529fe741292ff0b3977a01195bf0247425794
Python 3.13.13, Qt 6.11.1, Coin 4.0.8, Vtk 9.4.2, boost 1_86, Eigen3 3.4.1, PySide 6.11.1
shiboken 6.11.1, SMESH 7.7.1.0, xerces-c 3.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/35.2721/1.25

this bug did not exist in .19 i have a lot of files that were done in .19 they all worked doing this; they all broke after i opened them again in the above version; there are multiple  reports of this and all have been closed with no intended fix coming;

#16012 
#24451
and a few that were listed when i searched to find this report which seems to be the first one;

the problem came in release .20 and has never been fixed and is still in the software;

[unnamed.FCStd.zip](https://github.com/user-attachments/files/31238852/unnamed.FCStd.zip) this is a new file created in the above version; the problem is still very much there; the solution that is given name the constraint; that is only masking the problem and eventually this problem is going to surface and be a problem and naming the constraint will no longer resolve it;

i have been using, teaching and supporting 2D and 3D CAD systems since 1990; you so called experts developing the software need to be honest with yourselves and deal with the problems in this software if you want it to ever find its way into mainstream use;