FreeCAD/FreeCAD #7393: [Bug] Part tessellate function returns vastly different results when FreeCAD is in console mode

30350n, 2022-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

issue description is pretty clear, this is almost definitely a bug

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/Germany (en_DE)
Installed mods: 
  * Assembly4 0.12.3
  * ExtremeProDark 1.0.4
  * fasteners 0.4.2
  * fcgear 1.0.0
  * free2ki
  * kicadStepUpMod 10.14.5
  * sheetmetal 0.2.56
  * slic3r-tools
  * ThreadProfile 1.82.0
[/code]
```

### Subproject(s) affected?

Part

### Issue description

The `shape.tessellation(...)` function returns vastly different meshes, regarding resolution, depending on whether or not your run with or without the FreeCAD gui.

Running the attached script (`tessellation_bug.py`) on the attached example object like this:
`freecadcmd sphere.FCStd tessellation_bug.py` results in a mesh with `13360` vertices, while running it normally like this:
`freecad sphere.FCStd tessellation_bug.py` only results in a mesh with `2505` vertices.

My first thought was that this would be some kind of unit (metric vs. imperial) problem, but I wasn't able to find a constant to reproduce the "gui" version, from the command line.

From further testing, I figured that this mainly seems to affect circular objects, so my best guess is that it's somehow related to some kind of hidden angular deflection value.

### Anything else?

[tessellation_bug.zip](https://github.com/FreeCAD/FreeCAD/files/9416890/tessellation_bug.zip)
(contains an example FCStd file: `sphere.FCStd` and a python script which tessellates the object and prints the number of vertices: `tessellation_bug.py`)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-08-25:
Do you mind posting to the forum as well? Lots more traffic there. Just us triagers here.

30350n, 2022-08-25:
Done.

luzpaz, 2024-10-24:
Workaround: https://forum.freecad.org/viewtopic.php?p=639615&sid=14d1082986e3618d02fe59f582ae44e0#p639615

closed 2025-02-12 by github-actions[bot]