FreeCAD/FreeCAD #7395: [Feature Request] Unit system with dimensions displayed in meters

FEA-eng, 2022-08-25:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=71365

### Subproject(s) affected?

Core

### Idea description

The currently available MKS (m/kg/s/degree) unit system doesn't result in dimensions being displayed in meters - millimeters are still used everywhere. For some applications (e.g. civil engineering) it would be useful to have a unit system that actually changes the display of all dimensions to meters.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-08-25:
Looking at the code MKS is supposed to display values bellow 100mm in mm and values above that but bellow 10000000mm are displayed in m and so on for even higher or lower values
https://github.com/FreeCAD/FreeCAD/blob/a8cb4e744483ce2c4b0770b948105a766330a4b2/src/Base/UnitsSchemaMKS.cpp#L57-L64
These are sane ranges IMO, if there is somewhere where mm are being used regardless of these rules I think that would be a bug but we would need examples of that.

adrianinsaval, 2022-09-21:
@FEA-eng respond please, where is there an issue where mm are displayed with MKS? if no example is given this issue should be closed IMO, or probably closed either way and instead make an issue for the field that is not respecting the unit schema.

FEA-eng, 2022-09-21:
@adrianinsaval I would expect the (m/kg/s/degree) unit system to display meters at all times, not only when the dimensions are close to 1 m. That's how it's done in other CAD software (like SolidWorks). Seeing mm displayed everywhere (for example the default 10 mm depth of extrusion) with MKS system enabled is really confusing. Maybe this request should be changed to something like "make the MKS system display meters regardless of the dimension value" instead.

adrianinsaval, 2022-09-21:
I see no reason why you would want to see 10mm as 0.01m, "other software does it" is not a very good argument (although it can be valid for some stuff). Can you give me an example of a situation where forcing m for every value is advantageous?

FEA-eng, 2022-09-22:
Small dimensions displayed in meters may seem inconvenient but it's mainly about consistency (and clarity). Especially if you are going to show the dimensions to someone, then the units should be consistent according to the rules of engineering. I'm not a civil engineer but guys from that field would likely agree with me that the MKS system should show meters at all times.

adrianinsaval, 2023-05-05:
the problem with making MKS show in meters for values bellow 0.1m is that our default number of decimals is just 2 IIRC, meaning it'll get rounded a lot. Maybe we can make this range dependent on the configured number of decimals

joaocoutojc, 2023-08-02:
In freecad version 0.21 rc the measurements below ten meters show in millimeters. Could be shown in millimeters measured smaller than 0.1m

adrianinsaval, 2023-08-02:
> In freecad version 0.21 rc the measurements below ten meters show in millimeters. Could be shown in millimeters measured smaller than 0.1m

that would be when using the Standard unit system, MKS lowers this to values bellow 100mm as explained above.

furgo16, 2024-04-16:
> Small dimensions displayed in meters may seem inconvenient but it's mainly about consistency (and clarity). Especially if you are going to show the dimensions to someone, then the units should be consistent according to the rules of engineering.

I agree with @FEA-eng's point. It is less about precision, but about consistency and predictability. I was quite confused when I set units to be `m` and measurements where displayed in `mm`. I just learned about the unwritten rule of switching to `mm` below 1 `m` measurements by trying to file an issue and stumbling upon thid one. I would argue that if someone requires more precision below 1 `m`, they are already able to switch to the other `cm` or `mm`-based unit systems.

In addition, in FreeCAD 0.22 there is a new `Meter decimal (m, m², m³)` unit system that behaves under the expectations of this issue description. That is, measurements are always shown in `m`. I would also argue that to make those two unit systems (`MKS` and `Meter decimal`) also consistent with each other, units should always be in `m`.

FEA-eng, 2024-07-04:
Closing as completed by #11365

#5906 can stay open if there's a decision to do something with the inconsistent behavior of the MKS system in the future.

closed 2024-07-04 by FEA-eng