FreeCAD/FreeCAD #7400: [macOS] regression in 0.20.1

core-code, 2022-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=490819

### Version

0.21 (Development)

### Full version info

```shell
FreeCAD_0.20-1-2022-08-20-conda-macOS-x86_64-py310.dmg
```

### Issue description

a long standing problem in freecad's macOS integration was that is didn't have a CFBundleIdentifier, which is a prerequisite for any Mac app

this was finally fixed in 0.20, which finally had a proper bundle identifier: org.freecadteam.freecad

however, in 0.20.1 things are bad again, no bundle identifier. this is a clear regression.

### Anything else?

p.s. whats up with the random filenames changing pattern in each release? "FreeCAD_0.20-1-2022-08-20-conda-macOS-x86_64-py310.dmg" doesn't follow any pattern that previous releases had. every new release some new pattern is used. also the first "-" should definitely be a "." as its a "0.20.1" release...

p.p.s the CFBundleVersion is also damaged. it should be 0.20.1 not 0.20-1

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-08-27:
please use a more descriptive title for the issue

@looooo can you weight in on this?

adrianinsaval, 2022-08-27:
shouldn't the identifier be `org.freecad.FreeCAD`? or `org.freecadweb.FreeCAD`, the flatpak uses the latter and we ship the app icon with that identifier too.

looooo, 2022-08-28:
Bundles for 0.20.1 and weekly-builds should be updated soon:
https://github.com/FreeCAD/FreeCAD-Bundle/commit/d584c3ae76c2dcff353d0bf3ba99ce8f0b5ad83b
https://github.com/FreeCAD/FreeCAD-Bundle/commit/130bbd7d81b93899addfa8192df4620b4a84a5a6

luzpaz, 2022-09-12:
@looooo anything left here?

luzpaz, 2022-12-21:
@core-code please confirm fix

core-code, 2022-12-21:
@luzpaz would like to but there are no macOS binaries for 0.20.2

luzpaz, 2022-12-21:
Good to know but nevertheless @core-code can you check the 0.20.1 binaries though? (they should have the changes looo mentioned in https://github.com/FreeCAD/FreeCAD/issues/7400#issuecomment-1229426501

core-code, 2022-12-21:
the 0.20.1 binaries do NOT have a bundle identifier as mentioned in the initial top post here ( Aug 26 ·):

https://github.com/FreeCAD/FreeCAD/issues/7400#issue-1352448531

adrianinsaval, 2022-12-21:
try the bundle from here: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.20.1 I think the one in the main repo was not updated.

should this also be changed? https://github.com/FreeCAD/FreeCAD/blob/f0a4ec8240d45fe316c594f310dd28060600e666/src/MacAppBundle/FreeCAD.app/Contents/Info.plist#L63-L64

core-code, 2022-12-21:
now this is confusing

this one is broken (no bundle identifier)
https://github.com/FreeCAD/FreeCAD/releases/download/0.20.1/FreeCAD_0.20-1-2022-08-20-conda-macOS-x86_64-py310.dmg

this one is fine (bundle identifier = org.freecadweb.FreeCAD)
https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/0.20.1/FreeCAD_0.20.1-2022-08-28-conda-macOS-x86_64-py310.dmg

adrianinsaval, 2022-12-21:
Yeah the bundle was fixed but nobody remembered to also upload it on the main repo. The one I linked is done in a mostly automated way but has to be manually copied on the main repo.

luzpaz, 2023-01-14:
I think I took care of this (I've manually uploaded the the release page). Please test @core-code

core-code, 2023-01-14:
thanks! i can confirm that the newest release ( https://github.com/FreeCAD/FreeCAD/releases/download/0.20.2/FreeCAD_0.20.2-2022-12-27-conda-macOS-x86_64-py310.dmg ) has a proper bundle identifier ( org.freecadweb.FreeCAD ) and a version number ( SV: empty BV: 0.20.2-29603 ).

closed 2023-01-14 by core-code