FreeCAD/FreeCAD #7402: [Bug] <title>

claasahl, 2022-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

-

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4
```

### Subproject(s) affected?

Path

### Issue description

Not entirly sure if this a bug or an intentional change. In an earlier version of FreeCAD (see below), one was able to use the "Face"-tool on the face of a champfer to create a milling path which would approximate the face (in a sort of stair-like pattern - see picture). 

[image]

The attached [sample](https://github.com/FreeCAD/FreeCAD/files/9435028/sample.zip) was created with the following version.

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)
```

This functionality is no longer present in any of the (Windows) releases that followed. I opened the attached sample in the following versions and attempted to use the same tool ("Face"-tool) to create a similar / the same milling path, but no luck (i.e. no milling path is created). Was this change intended? If so, how would one create a similar milling path in the latest version (0.20 or 0.20.1)?

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4
```

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-08-27:
Created discussion thread: https://forum.freecadweb.org/viewtopic.php?f=15&t=71436

claasahl, 2022-08-28:
I had hoped the "3D Pocket"-tool would produce a comparable result, but (at least in my case) it produces no milling path at all when selecting the chamfer's face and adjusting the step down (depth).

[image]

For what it is worth, I had a look at other tools in v0.20 (i.e. "Profile", "Pocket Shape", "Drilling", "Face", "Helix", "Adaptive", "Engrave", "Deburr" and "3D Pocket"). None of them produced a result comparable to the previously shared example FCStd-file. To be fair, most of the tools shouldn't as they are meant for something completely different. That said, I had expected one of "Profile", "Pocket Shape" or "3D Pocket" to do the trick.

luzpaz, 2022-08-28:
@claasahl please participate in the forum thread

claasahl, 2022-08-29:
@luzpaz Is the break in communication medium really necessary? I prefer not to signup for a forum for a single discussion.

My intention was to share my observation with the means to reproduce them. If there are any follow up questions I'll gladly answer them here.

luzpaz, 2022-08-31:
Path is a very dynamic workbench development-wise. Most path users are using the development version of FreeCAD. Not many path fixes and code gets backported to the stable branch and therefore you may be reporting obsolete tickets. It's more of a triage thing for us. We traditionally have run bugs through the forum before posting them or else we get inundated by tickets and obsolete ones (since they've been fixed in dev version).

claasahl, 2022-09-01:
Thanks for the explanation. I did give the development version a try (`0.20.1.29410` as listed in the description). I simply choose to select the latest release as that felt more appropriate (rather than highlighting this for a version that is probably undergoing constant changes).

luzpaz, 2023-01-08:
@claasahl is this still the case for v0.20.2 ?

claasahl, 2023-01-28:
@luzpaz apologies for the late reply. yes, the issue remains in version:

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4
```

luzpaz, 2023-01-29:
@mlampert do you have a moment to weigh-in ?

mlampert, 2023-01-30:
I can't verify that this ever worked and as mentioned before the path in the sample file doesn't look like anything the Path WB would generate.

I don't think this is a bug, the feature is not something the Facing operation should be able to handle. Other than the 3d operations all Ops currently work on 2.5d basis. Having said that, the feature _can_ be milled with the 3d ops available including, as Gene pointed out, the Pocket3D.

If there is a bug here it is that seemingly the Facing op ever processed this face.

closed 2025-01-04 by github-actions[bot]