FreeCAD/FreeCAD #7410: [Bug] Output file naming issues

dresco, 2022-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=71059

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Fedora Linux 35 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.30143 (Git)
Build type: Release
Branch: master
Hash: a72bf218f9917df37e811d7e73a36333282028f5
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.12.3
  * FeedsAndSpeeds 0.4.0
  * kicadStepUpMod 10.10.4
[/code]
```

### Subproject(s) affected?

Path

### Issue description

Description from forum post;

Recent master is showing multiple defects in naming GCODE output file.

1. cancelling file save dlg throws about ten lines of red letter errors in report view.
2. hover hint inconsistent. It says " /home/cnc/%d.g-code" will store file in /home/freecad !
3. %d not processed, it comes out as %d in the file name.
4. any supplied suffix like the suggested ".g-code" gets lost and proposed filename has no extension.
5. Hitting save btn does nothing on first attempt, Needs two clicks .

### Anything else?

This appears to resolve the save dialog issues, and the errors on cancel..

Note that the Activated() function was previously comparing the first returned parameter from exportObjectsWith() as True/False, but is now checking as None - which is what the file name is set to - rather than the return value. Have changed it to use the name, assuming that was the intent?

Happy to submit a PR if in agreement with these changes (and can address the hover hint also)..

```
diff --git a/usr/Mod/Path/PathScripts/PathPost.py b/usr/Mod/Path/PathScripts/PathPost.py
index f185678..527278a 100644
--- a/usr/Mod/Path/PathScripts/PathPost.py
+++ b/usr/Mod/Path/PathScripts/PathPost.py
@@ -238,19 +238,21 @@ def resolveFileName(job, subpartname, sequencenumber):
 
     if openDialog:
         foo = QtGui.QFileDialog.getSaveFileName(
-            QtGui.QApplication.activeWindow(), "Output File", filename
+            QtGui.QApplication.activeWindow(), "Output File", fullPath
         )
         if foo[0]:
             fullPath = foo[0]
         else:
             fullPath = None
 
-    # remove any unused substitution strings:
-    for s in validPathSubstitutions + validFilenameSubstitutions:
-        fullPath = fullPath.replace(f"%{s}", "")
+    if fullPath:
+        # remove any unused substitution strings:
+        for s in validPathSubstitutions + validFilenameSubstitutions:
+            fullPath = fullPath.replace(f"%{s}", "")
+
+        fullPath = os.path.normpath(fullPath)
+        PathLog.track(fullPath)
 
-    fullPath = os.path.normpath(fullPath)
-    PathLog.track(fullPath)
     return fullPath
 
 
@@ -591,7 +593,7 @@ class CommandPathPost:
             filenames.append(name)
             PathLog.track(result, gcode, name)
 
-            if result is None:
+            if name is None:
                 success = False
             else:
                 finalgcode += gcode
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-08-28:
Thank so much for the thorough report. Assigning to sliptonic for review.

closed 2022-09-03 by sliptonic

donovaly, 2022-09-12:
was the bug also in FreeCAD 0.20.x? If yes, @sliptonic , should I backport or could you do this?

PR #7428 (2022-09-03): Path: Fix for gcode output file naming issues.
Use the fullPath name when opening Output File dialog, handle cancel cleanly.

These changes fix output file file naming issues. Without them, the file dialog
shows the unprocessed substitution strings in the current working directory,
and generates errors in report view on Cancel.

Fixes #7410 

Thank you for creating a pull request to contribute to FreeCAD! To ease integration, we ask you to conform to the following items. Pull requests which don't satisfy all the items below might be rejected. If you are in doubt with any of the items below, do [... cut]