FreeCAD/FreeCAD #7418: [Bug] <title>in mac before changes lang. can't open Preferences ->General<title>

wisechien, 2022-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

:Bug reports

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

in mac before changes lang. can't open Preferences ->General

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-08-30:
Please open a forum thread.

luzpaz, 2022-09-09:
Asked @wisechien to open a forum thread. Didn't happen. Closing ticket.

closed 2022-09-09 by luzpaz