FreeCAD/FreeCAD #7421: [Bug] `freecad -c` throws `t.qpa.xcb: could not connect to display ..` when run from Docker > PHP > Laravel stack

agustavo87, 2022-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=620415

### Version

0.20 (Release)

### Full version info

```shell
FreeCad AppImage at https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/0.20/FreeCAD_0.20-29177-2022-06-15-conda-Linux-x86_64-py39.AppImage
```

### Issue description

Run from Laravel Sail docker container on a [Laravel 8.0 Application from Windows Subsystem for Linux (WSL)](https://laravel.com/docs/8.x#getting-started-on-windows).  
```sh
composer create-project laravel/laravel:^8.0 freecad-issue
cd freecad-issue
composer require laravel/sail --dev
# The next could take a while, no service is necessary
php artisan sail:install 
./vendor/bin/sail up
# Laravel app should be accessible on http://localhost

# Publish Dockerfiles on docker/8.1/Dockerfile
sail artisan sail:publish

# Download AppImage
wget -O freecad.AppImage https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/0.20/FreeCAD_0.20-29177-2022-06-15-conda-Linux-x86_64-py39.AppImage
chmod +x freecad.AppImage
```

Modify `docker/8.1/Dockerfile` at the end to add freecad dependencies

```Dockerfile
RUN apt-get clean && apt-get update \
    && apt-get install -y libgl1
ENV APPIMAGE_EXTRACT_AND_RUN=1 
```

Then stop, rebuild, and restart image
```sh
./vendor/bin/sail down
./vendor/bin/sail build
./vendor/bin/sail up -d
```

modify `.env` file at the end
```env
APPIMAGE_EXTRACT_AND_RUN=1 
```

Now  test if the AppImage is working
```sh
./vendor/bin/sail shell
./freecad.AppImage -c "print(2+2)" #4
```

If is working as expected the web route can be added to check the freecad runned from a HTTP request. On `routes/web.php` add at the end:
```php
Route::get('/freecad', function () {
 return 'result:'.`../freecad.AppImage -c "print(2+2)" 2>&1`;
});
```
If at the browser one go to `localhost/freecad` it show
```
result:qt.qpa.xcb: could not connect to display qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found. This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem. Available platform plugins are: eglfs, minimal, minimalegl, offscreen, vnc, webgl, xcb. /tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/AppRun: line 42: 65 Aborted ${MAIN} "$@"
```

### Anything else?

I have tried to isolate the issue and have found the next.
- All this is tested inside Sails Docker container: https://github.com/laravel/sail/tree/1.x/runtimes/8.1
- First, I didn't have problems running other commands from inside Laravel from a HTTP request.
- The thing works OK, with php alone or[ php built-in server.](https://www.php.net/manual/en/features.commandline.webserver.php)
- The problem happens when the artisan server is run from the "sail" Linux (docker container) user without "no-reload" option ([https://laravel.com/docs/8.x/upgrade#ph ... oad-option](https://laravel.com/docs/8.x/upgrade#php-artisan-serve-no-reload-option)), if the option is  used, the things works ok. Also, if the artisan server is run from "root" the problem does not happen, even without the "no-reload" option.
- If the artisan server is run from supervisor as "sail" user the problem happens (even without "no-reload" option), but without user specification (as root?) does not.

Right now, is solved (at least temporarily) running things vía a script with "no-reload" option as sail user, but this need to be run manually, and can't use supervisor. But this don't clarify what could be happening to make freecad run correctly as headless in some situations, and try to run GUI mode in the others

## Workaround
To this is necessary to disable supervisor running artisan server. On `docker/8.1/supervisor.conf`
```conf
[supervisord]
nodaemon=true
user=root
logfile=/var/log/supervisor/supervisord.log
pidfile=/var/run/supervisord.pid

# [program:php]
# command=/usr/bin/php -d variables_order=EGPCS /var/www/html/artisan serve --host=0.0.0.0 --port=80
# user=sail
# environment=LARAVEL_SAIL="1"
# stdout_logfile=/dev/stdout
# stdout_logfile_maxbytes=0
# stderr_logfile=/dev/stderr
# stderr_logfile_maxbytes=0
```
Then the image can be rebuild
```sh
./vendor/bin/sail down
./vendor/bin/sail build
./vendor/bin/sail up -d
```
Now the web app is not automatically seved by supervisor `http://localhost` should give error. So the server can be started manually and some tweaks can be made, for example using the `--no-reload` option.
```sh
./vendor/bin/sail shell
/usr/bin/php -d variables_order=EGPCS /var/www/html/artisan serve --host=0.0.0.0 --port=80 --no-reload
```
Now, for some reason, `http://localhost/freecad` shows as expected `result:4`. Other tweaks that make things works is running from `sail root-shell` or disabling only `# user=sail` on `supervisor.conf`.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

agustavo87, 2022-09-05:
Tried to run directly some parts of FreeCAD with launching options from AppImage as sugested in the GH [issue](https://github.com/FreeCAD/FreeCAD-Bundle/issues/30#issuecomment-583720252) :
```sh
./FreeCAD.AppImage
./FreeCAD.AppImage freecadcmd
./FreeCAD.AppImage python --version
./FreeCAD.AppImage python
```
The results where the next
- `freecad.AppImage freecadcmd "print(2+2)”`:
  - from Shell
Version info. (no `4` returned)
  - from HTTP
```
Initialization of FreeCAD failed: While initializing FreeCAD the following exception occurred: 'Could not create directories. Failed with: Permission denied' Python is searching for its runtime files in the following directories: /tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr/lib/python39.zip:/tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr/lib/python3.9:/tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr/lib/python3.9/lib-dynload Python version information: 3.9.13 | packaged by conda-forge | (main, May 27 2022, 17:07:55) [GCC 10.3.0] The environment variable PYTHONHOME is set to '/tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr'. Setting this environment variable might cause Python to fail. Please contact your administrator to unset it on your system
```
- `freecad.AppImage python -c "print(2+2)"`:
  - from Shell
`4`
  - from HTTP
`4`
- `freecad.AppImage freecadcmd -c "print(2+2)"`:
  - from Shell
`4`
  - from HTTP
```
Initialization of FreeCAD failed: While initializing FreeCAD the following exception occurred: 'Could not create directories. Failed with: Permission denied' Python is searching for its runtime files in the following directories: /tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr/lib/python39.zip:/tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr/lib/python3.9:/tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr/lib/python3.9/lib-dynload Python version information: 3.9.13 | packaged by conda-forge | (main, May 27 2022, 17:07:55) [GCC 10.3.0] The environment variable PYTHONHOME is set to '/tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr'. Setting this environment variable might cause Python to fail. Please contact your administrator to unset it on your system.
```

Apparently, when launching `python`, even inside the `AppImage` and above the `docker` > `php` > `php built-in server` > `laravel artisan serve` stack everything seems to be working as expected, but when calling `freecadcmd` something happens. But this is an interaction with the described stack, becouse from `docker` > `shell` seems to be working as expected.

adrianinsaval, 2022-09-21:
this is expected behavior IMO since the GUI is not available there, freecadcmd should be used instead, in the forum discussion another problem arises from this so either this issue should be updated or better closed IMO.

agustavo87, 2022-09-27:
> this is expected behavior IMO since the GUI is not available there

So why does this work from shell? Also works when the HTTP is handled by a server (i.e., PHP built-in server) that is run as root.
Also tried with `freecadcmd` from inside AppImage  ([forum discussion](https://forum.freecadweb.org/viewtopic.php?p=623775#p623775))

`freecad.AppImage freecadcmd -c "print(2+2)"`
- from Shell
`4`
- from HTTP
```
Initialization of FreeCAD failed: While initializing FreeCAD the following exception occurred: 
'Could not create directories. Failed with: 
Permission denied' Python is searching for its runtime files in the following directories:
/tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr/lib/python39.zip:/tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr/lib/python3.9:/tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr/lib/python3.9/lib-dynload 
Python version information: 3.9.13 | packaged by conda-forge | (main, May 27 2022, 17:07:55) [GCC 10.3.0] 
The environment variable PYTHONHOME is set to '/tmp/appimage_extracted_8cac30c456e0f6469e8601386557085d/usr'. Setting this environment variable might cause Python to fail. Please contact your administrator to unset it on your system.
```

adrianinsaval, 2022-09-27:
Yeah but that looks like a different problem and it should be treated as such.

luzpaz, 2022-10-03:
Does that merit opening a ticket on [FreeCAD-bundle](https://github.com/FreeCAD/FreeCAD-Bundle/)? (where appimage related bugs are filed)

adrianinsaval, 2022-10-03:
Not IMO, the GUI issue seems to be a problem with the environment and same thing with the permission issue above, in the forum thread that was solved too but FreeCAD still fails. It's not clear if this is really related to being appimage, since it only fails on that environment I would think not. OP hasn't replied with a log file yet so not much can be done.

luzpaz, 2022-10-16:
@agustavo87 ping

luzpaz, 2023-03-15:
@amrit3701 is this reproducible with your docker project ?

amrit3701, 2023-03-15:
@agustavo87 

If you want to run FreeCAD code on server, then I would suggest to use [FreeCAD CLI docker images](https://hub.docker.com/r/amrit3701/freecad-cli/). 

Here is the docs: [https://wiki.freecad.org/FreeCAD_Docker_CLI_mode](https://wiki.freecad.org/FreeCAD_Docker_CLI_mode)

In FreeCAD docker images, FreeCAD is compiled without GUI element and you will never display error unless that feature require GUI elements.

agustavo87, 2023-04-03:
> OP hasn't replied with a log file yet so not much can be done.

- [https://forum.freecad.org/viewtopic.php?p=624050#p624050](https://forum.freecad.org/viewtopic.php?p=624050#p624050) ?
- [https://forum.freecad.org/viewtopic.php?p=632887&sid=a3acc667589c6e657ccedcb375fb1627#p632887](https://forum.freecad.org/viewtopic.php?p=632887&sid=a3acc667589c6e657ccedcb375fb1627#p632887) ?

adrianinsaval, 2023-04-07:
have you tried a recent development version? can you still reproduce?
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Can you also reproduce using freecad packaged some other way? (snap, flatpak or system pakcage)

maxwxyz, 2024-06-13:
Is this still relevant?

luzpaz, 2024-07-27:
@agustavo87 please respond

adrianinsaval, 2024-11-05:
no relevant activity since 2023 so I think it's not worth keeping this open.