FreeCAD/FreeCAD #7425: boolean from a sweep doesn't work

happypascal, 2022-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=71324

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * Curves 0.4.4
[/code]
```

### Issue description

In PartWB, I created a sweep along an helix with Frenet set to TRUE.
Then I tried to create a union of this sweep with another shape but the resulting shape was invalid.

Same problem with difference or common boolean operation. Same with Frenet to FALSE.

Instead of using the helix from PartWB, I created an helix by mapping a drawing on a round surface.
The resulting shape was not invalid but it doesn't work either.

[_____test.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/9468633/_____test.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-09-09:
Have you checked other sweep bugs in the bugtracker?

happypascal, 2022-09-09:
> Have you checked other sweep bugs in the bugtracker?

Yes I search issues related to WBpart. I found 2 bugs with boolean but none related to an helix part.

PLChris, 2022-09-10:
User error: self intersection of helix. Pitch too small or section too high.

closed 2022-09-10 by PLChris

happypascal, 2022-09-10:
> User error: self intersection of helix. Pitch too small or section too high.

Reduced section height by 1% to avoid intersection. Boolean works like a charm. Sorry for the disturbance. Thanks