FreeCAD/FreeCAD #7430: [Feature Request] Local Coordinate System Placement: Detached with 3 Points

bernard01, 2022-09-02:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=20&t=71521

### Subproject(s) affected?

PartDesign

### Idea description

The objective is to create a Local Coordinate System (LCS) which cannot be attached to any vertex in the drawing.

This is very fundamental and it should be supported in a way as streamlined and user-friendly as possible.
It is more critical with the use of the Assembly workbench because it is used quite often with it.

I have tried various ways to achieve it, and yes, it is possible with the currently available tools, but very cumbersome.

Let me first describe a specific case, which is as simple as it gets because all 3 LCS axes are aligned with the CS in the body's origin. Please refer to the attached screen shot.

I know I want a LCS such as:
X axis must be in -X axis direction of base CS
Y axis must be in +Z axis direction of base CS
Z axis must be in +Y axis direction of base CS

To get there, I need to get into the Placement dialog which is away from the Local Coordinate System Parameters dialog which is entered through the Properties panel.

After some trial and error, I get my LCS With Euler angles:
Yaw 180.00 °
Pitch 0.00 °
Roll 90.00 °

These are actually 2 rotations:
180 around Z gets X in place
90 around X gets both Y and Z axes in place.

You see these are 2 rotations that need to be done to transform these two coordinate systems one into the other (using Euler angles). The problem I have is that I cannot keep these 3 coordinate systems in my head. I have to keep them in my head because even when I do one step at the time, I cannot see the result in the visualization in the main window. Because the LCS has all 3 axes IN GREEN COLOR because it is selected. So I don't see where X, Y and Z axes are moving when I play with the angles. I have to first de-select the LCS but that is not an option because it cannot be done without losing the user interface for making the changes.

Obviously this back-to-front trial and error approach is not what I want.

What I want is the following, and I know it works. I know it works from two perspectives:
1) I have used the following method in a different 3D CAD modeler
2) I have verified the following for FreeCAD using the example above.

At the minimum, I need the ability to input 3 points that define a rotation matrix as follows:

double[][] matrix = new double[][]{
    {-1, 0, 0},// Local X axis
    {0, 0, 1},// Local Y axis
    {0, 1, 0},// Local Z axis
};

I gleaned the method from:

[Rotation](https://commons.apache.org/proper/commons-math/javadocs/api-3.6.1/org/apache/commons/math3/geometry/euclidean/threed/Rotation.html)

"Build a rotation from a 3X3 matrix"

Data entry of 3 points, relative to the origin of the LCS. As simple as that.

Currently, I use a little Java program that converts these 3 points into Euler angles.

I am not saying that the proposed method is what it should be. I am saying that it should just be a few mouse clicks without much thinking to get a detached LCS. It has been said that there are some 90 degree buttons that can be clicked until the desired outcome is achieved, but I cannot see any such buttons in Part Design Workbench, and I cannot see a way to move the LCS axes around in a truly interactive way.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bernard01, 2022-09-15:
Not an issue, found better existing solution

closed 2022-09-15 by bernard01

luzpaz, 2022-09-21:
@bernard01 please elucidate (for posterity)

bernard01, 2022-09-25:
> 

I found it best to attach to a plane of the Origin LCS then flip and rotate. That is easy enough to visualize. There are more then enough ways of doing things in FreeCAD I think. My request comes from experience with an older CAD program that had a different way of specifying the LCSs as triplets of vectors in their respective origins.