FreeCAD/FreeCAD #7439: [Bug] 3D Surface on complex geometry won't finish if set to BaseBoundBox

pathfinder555, 2022-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=67696

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.30256 (Git) AppImage
Build type: Release
Branch: master
Hash: 55866fd62221beadad3cfd60915278b156609a66
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Path

### Issue description

When trying to use the 3D surfacing operation on complex objects (In my case specifically imported and converted mesh objects with a lot of small faces), if the BoundBox setting is set to "BaseBoundBox"(i.e. the default), the operation won't finish and the program will hang. Changing the option to "Stock" solves the issue. 

A temporary fix might be to change the default for the Operation to "Stock" as default, to prevent forgetful and/or hasty people (like me) from losing their unsaved progress.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GeneGH, 2022-09-04:
I have seen this sort of behavior at times, but not consistently. I just ran a simple trial and the times were virtually identical for stock and base bound box. (Small file, about 30 seconds each on my computer.)

I find that the Mesh Conversion parameters (Angular Deflection and Linear Deflection) are extremely important. Even modest changes can make a huge difference in the time for the process to complete. I have not determined if this has any relationship to the bounding box.

I tried running the mesh file referenced above, and I can confirm the major change in process time. The mesh object has a few errors in it, but I do not know that would make any difference in the bounding box behavior.

[code]
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.21.30143 (Git)
Build type: Release
Branch: master
Hash: a72bf218f9917df37e811d7e73a36333282028f5
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.4.0
  * FeedsAndSpeeds-master.zip
  * Help 1.0.3
  * offline-documentation
[/code]

Gene

pathfinder555, 2022-09-04:
> 
> I tried running the mesh file referenced above, and I can confirm the major change in process time. The mesh object has a few errors in it, but I do not know that would make any difference in the bounding box behavior.
> 
>OS: Windows 7 Version 6.1 (Build 7601: SP 1) Word size of FreeCAD: 64-bit Version: 0.21.30143 (Git) Build type: Release Branch: master Hash: [a72bf21](https://github.com/FreeCAD/FreeCAD/commit/a72bf218f9917df37e811d7e73a36333282028f5) Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2 Locale: English/United States (en_US) Installed mods:
>
 >   * FeedsAndSpeeds 0.4.0
>
  >  * FeedsAndSpeeds-master.zip
>
 >   * Help 1.0.3
>
>   * offline-documentation
> 
> 
> Gene

Were you able to get a result at all? Because I haven't. To investigate further, I tried a smaller cutout of the file above (about a quarter of its size; the original file and the cutout were both made in Blender). Using Stock, I had a processing time of 0.91117 sec. Using BaseBoundBox, it took around 50 minutes to get to a result and at some point during the computation, FC started to throw a ton of "<Path.Area> Area.cpp(992): failed to close some projection wire" and "<Path.Area> Area.cpp(452): ccurve not closed" errors, then, after a few more minutes ended the computation with this message:

[code]
<Path.Area> Area.cpp(1633): skipped 30 sub shapes during projection
Failed to pre-process base as a whole.
3D Surface operation time is 49 min. 12.25031 sec.
[/code]

The file in question: https://we.tl/t-2JlJUEKlVb

sliptonic, 2022-09-05:
The input mesh is messy with some flipped normals and duplicate faces.  Correcting that doesn't seem to affect it.  I can duplicate the problem.

This looks like something inside the OCL surface operation.  
I don't have any problem with changing the default to 'Stock' for the Boundbox setting.  That doesn't fix the problem but avoids it somewhat.  Interestingly, setting the stock size to exactly match the shape boundbox still works.

The sample shape has some interesting characteristics like thru-holes and edge features.  It would be interesting to try to narrow down exactly which features start triggering the problem.

pathfinder555, 2022-09-06:
> Interestingly, setting the stock size to exactly match the shape boundbox still works.

Well if that's the case, wouldn't the simplest (but admittedly hackiest) solution be to essentially fake the Boundbox by creating a stock that has the dimensions of the boundbox in the background and just not use the real boundbox function at all?

luzpaz, 2023-09-23:
@pathfinder555 the file is expired. Can you upload it to github ? or a dropbox type of thing ?

luzpaz, 2024-08-27:
@spanner888 do you have the ability/time to confirm the diagnosis of this issue ?

spanner888, 2024-08-27:
At present not enough of ability or time, but thanks for asking.

luzpaz, 2024-09-07:
bumping for visibility

closed 2025-01-12 by github-actions[bot]