FreeCAD/FreeCAD #7450: [Bug] Model tree view not always updating object groups when scripting

goatchurchprime, 2022-09-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

.

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: NixOS 22.11 (Raccoon) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.20.1.Unknown
Build type: Release
Python 3.10.5, Qt 5.15.5, Coin 4.0.1, Vtk 8.2.0, OCC 7.5.1
Locale: English/United States (en_US)
[/code]
```

### Issue description

Create a new empty document, and then paste the following 3 lines into the Python console

```Python
doc = App.ActiveDocument
g = doc.addObject("App::DocumentObjectGroup", "groupfolder")
g.addObject(doc.addObject("Sketcher::SketchObject", "pp2peer"))

```
This makes a group and puts a sketch into it, like so:
[image]

Now copy-paste the following two lines into the console to delete and create the object back in the group
```Python
doc.removeObject("pp2peer")
g.addObject(doc.addObject("Sketcher::SketchObject", "pp2peer"))
1;

```

After doing this a couple of times (**make sure you copy the final linefeed so both lines run at once**) the pp2peer object appears not to get moved into the groupfolder group.

[image]

But it is actually there internally, which you can see if you print `g.OutList`.

There does not seem to be a function I can call to force this Group tree node to regenerate so it matches the underlying document model, however if I right click on groupfolder and select "Create group...", the addition of the new group does make the node regenerate.

### Anything else?

Although this looks like quite minor, there is a danger it goes unnoticed and causes big problems to other macro developers who could spend hours trying to chase up a bug in a macro that isn't there because the display representation of the folders is wrong.  

I don't know how the node regeneration could be getting out of sync (sometimes called before or after the `g.addObject()`).  Maybe the `doc.recompute()` function could check and repair any errors at least as a stopgap measure.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Zolko-123, 2022-10-28:
did you try `Gui.updateGui()` ?

luzpaz, 2022-11-09:
@goatchurchprime ping

luzpaz, 2022-12-29:
@goatchurchprime :point_up_2:

Syres916, 2023-02-06:
It appears to me that the syntax has changed not sure exactly when in the dev cycle but this works as the OP would expect (note the extra `adjustRelativeLinks`) the `Gui.updateGui()` has no effect from my testing:

```
doc = App.ActiveDocument
g = doc.addObject("App::DocumentObjectGroup", "groupfolder")
g.addObject(doc.addObject("Sketcher::SketchObject", "pp2peer"))

g.removeObject(doc.getObject("pp2peer"))
doc.getObject('pp2peer').adjustRelativeLinks(g)
g.addObject(doc.getObject("pp2peer"))

```

Roy-043, 2023-10-05:
Closing as resolved.

Roy-043, 2024-02-23:
The issue has not been resolved after all. The original code has `doc.removeObject()`, Syres's code has `g.removeObject()`. Same function name, but different objects (document vs. group).

onekk, 2024-02-23:
Why not this way?

```
doc = App.ActiveDocument
g = doc.addObject("App::DocumentObjectGroup", "groupfolder")
g.addObject(doc.addObject("Sketcher::SketchObject", "pp2peer"))
doc.recompute()
```

It seems to work all the times, and rpobably is the way to go, as you can create multiple things and then invoke only one time the recompute() as it has some time cost.

Kind Regards

Carlo D.

Roy-043, 2024-02-23:
I would say your code avoids the issue, but does not solve it.

cfunseth, 2024-02-23:
> Why not this way?
> 
> ```
> doc = App.ActiveDocument
> g = doc.addObject("App::DocumentObjectGroup", "groupfolder")
> g.addObject(doc.addObject("Sketcher::SketchObject", "pp2peer"))
> doc.removeObject("pp2peer")
> g.addObject(doc.addObject("Sketcher::SketchObject", "pp2peer"))
> # maybe adding
> doc.recompute()
> ```
> 
> No problem with this code at least with this version:

Your code also exhibits the same results as the original code posted in the issue. The bug occurs when you run the execute the commands in a script and not one by one. For example, if you split up your code into sections like this:

```
doc = App.ActiveDocument
g = doc.addObject("App::DocumentObjectGroup", "groupfolder")
g.addObject(doc.addObject("Sketcher::SketchObject", "pp2peer"))
```

...and then paste the following code (including the "1;") into the Python console then you will end up with the sketch appearing outside of the group. (You may have to try it a couple times, but I just made it happen after two tries on my system)
```
doc.removeObject("pp2peer")
g.addObject(doc.addObject("Sketcher::SketchObject", "pp2peer"))
1;
```

cfunseth, 2024-02-23:
> The issue has not been resolved after all. The original code has `doc.removeObject()`, Syres's code has `g.removeObject()`. Same function name, but different objects (document vs. group).

@Roy-043 I know you're probably aware of this, but I wanted to comment on it as others (like me) may not be. Syres's code utilizing 'g.removeObject()' only removes it from the group. My use case (and I assume the OP's use case) is that I need to completely delete the original object and add a new one with the same name to the same group. I believe that means I need to use the 'document' version of removeObject(), which is a part of this overall issue.

Is there anything else I could provide context on for this? (I'm not a programmer, mainly just a consumer of FreeCAD!)

cfunseth, 2024-02-23:
One more comment:
I saw that there was also an 'object' level function called 'removeObjectsFromDocument()'. I attempted to use this instead of the 'document' level 'removeObject()' function, but the same result of the new object appearing outside the group in the tree view was encountered. 

Not sure if these functions are related in any way, but wanted to point it out as they appear in two separate parts of the code.

Roy-043, 2024-02-24:
> Is there anything else I could provide context on for this?

Not that I am aware of.

Roy-043, 2024-12-14:
Can't reproduce the issue in the version listed below. Closing this.
```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

closed 2024-12-14 by Roy-043