FreeCAD/FreeCAD #7451: [Feature Request] Please add exported file to "recent files"

AndKe, 2022-09-06:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=71675

### Subproject(s) affected?

Other (specify in description)

### Idea description

Let's say I export a 3mf file.
The next, very likely thing would be to open it in another tool.

It would be nice that the recently exported file shows up in "recent" file list (of file manager) - but it does not.
(Tested on Ubuntu)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

onekk, 2022-09-07:
I think that for Linux this link should be of some help.

https://specifications.freedesktop.org/recent-file-spec/recent-file-spec-latest.html

luzpaz, 2022-11-18:
Directly contradicts #6076

Roy-043, 2023-10-05:
As mentioned in #6076 there are hidden preferences for this.
Wiki documentation: https://wiki.freecadweb.org/Fine-tuning

closed 2023-10-05 by Roy-043

luzpaz, 2023-10-05:
How do we inform end-users of fine-tuning preferences?

Roy-043, 2023-10-07:
I'll add a note to the [Preferences Editor](https://wiki.freecad.org/Preferences_Editor) page.