FreeCAD/FreeCAD #7458: [Bug] Toolbars visually distorted on Fedora 35

tkontogi, 2022-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=555883#p555883

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Fedora Linux 35 (Thirty Five) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.Unknown
Build type: Unknown
Python 3.10.5, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.3, OCC 7.5.0
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * CurvedShapes 1.0.4
  * WebTools
  * Rocket 2.7.1
  * CADExchanger
  * slic3r-tools
[/code]
```

### Issue description

All tool bars are visually distorted and the look and feel also looks broken.

Please review attached images (from the Sketcher, but the same look and feel appears to all toolbox toolbars).

[image]
[image]
[image]

### Anything else?

100% reproducible
No log file is created. Happy to help you to help me on that.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-09-07:
Forum thread? (the one above is unrelated)

tkontogi, 2022-09-08:
Thank you.
Posted in the forum.
Updated this case with forum URL.

Appreciated.

tkontogi, 2022-09-08:
Issue related to DPI value, configurable in xfce4-appearence-settings --> Fonts

In my case the value was set to 95.
Setting it back to 96 solved the issue.

Solution provided in:  [https://forum.freecadweb.org/viewtopic.php?p=624718#p624718](url)

Thank you.

[image]

luzpaz, 2022-09-08:
Thanks for the follow-up. Closing!

closed 2022-09-08 by luzpaz