FreeCAD/FreeCAD #7464: [Bug] Menu bar

1959didier, 2022-09-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/

### Version

0.20 (Release)

### Full version info

```shell
The menu bar is half displayed on my Samsung 27" screen, version 0.20 and 0.21 of Freecad
```

### Issue description

Displaying the menus.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-09-09:
Thank you for the impulse to make a bug report. But our guidelines ask to first open a forum thread. This may be a DPI issue, see #7458

adrianinsaval, 2022-09-21:
full version information please, if OP does not follow up this issue should be closed.

luzpaz, 2022-09-22:
No response. Closing

closed 2022-09-22 by luzpaz