FreeCAD/FreeCAD #7468: [Bug] Test PathTests.TestPathDrillable.TestPathDrillable fails on s390x

coldtobi, 2022-09-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=71764

### Version

0.20 (Release)

### Full version info

```shell
Debian packaged version 0.20.1 -- Debian version 0.20.1+dfsg1-1

This information is from my amd64 machine, not from s390x, but versions should be the same.

[code]
OS: Debian GNU/Linux bookworm/sid 
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

Path

### Issue description

Hi,

I'm one of the people looking after FreeCAD in Debian, I've encountered a test suite error in the TestPathDrillable testsuite on the s390x Architecture.
The other architectures seems to be fine, so I'm quite puzzled about the reason behind.

s390x is for Debian a release architecture, so this failure currently prevents freecad to migrate to "testing", the staging area for the next release of Debian.

The failure happens on the Debian CI, the full log for (the failing) s390x can be found here: [https://ci.debian.net/data/autopkgtest/ ... 237/log.gz](https://ci.debian.net/data/autopkgtest/testing/s390x/f/freecad/25840237/log.gz), here the excerpt of the failing test:

```
======================================================================
FAIL: test10 (PathTests.TestPathDrillable.TestPathDrillable)
Test isDrillable
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/share/freecad/Mod/Path/PathTests/TestPathDrillable.py", line 78, in test10
    self.assertTrue(drillableLib.isDrillable(self.obj.Shape, candidate))
AssertionError: False is not true

======================================================================
FAIL: test20 (PathTests.TestPathDrillable.TestPathDrillable)
Test getDrillableTargets
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/share/freecad/Mod/Path/PathTests/TestPathDrillable.py", line 305, in test20
    self.assertEqual(len(results), 20)
AssertionError: 21 != 20

----------------------------------------------------------------------
Ran 736 tests in 46.072s

FAILED (failures=2)

```
As said, I have no idea where to start looking to pin point the problem, let me know if I can somehow provide more information.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-19:
@coldtobi is this something you can also reproduce on development version ?

luzpaz, 2023-02-19:
@coldtobi ping

coldtobi, 2023-02-21:
@luzpaz  thanks for the ping, I acutally forgot to follow up…

s390x situation is unfortunately worse now, as there are no [Qt5WebEngineWidgets for s390x](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=913043), so I had to configure freecad with ` -DBUILD_WEB=OFF -DBUILD_START=OFF` to get it compiling.

FTR, I was compiling 381f534c477b08148f956ac47e8341e63d3ac432 on the s390x porterbox.

```
======================================================================
FAIL: test10 (PathTests.TestPathDrillable.TestPathDrillable.test10)
Test isDrillable
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/tobi/FreeCAD/build/Mod/Path/PathTests/TestPathDrillable.py", line 77, in test10
    self.assertTrue(Drillable.isDrillable(self.obj.Shape, candidate))
AssertionError: False is not true

======================================================================
FAIL: test20 (PathTests.TestPathDrillable.TestPathDrillable.test20)
Test getDrillableTargets
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/tobi/FreeCAD/build/Mod/Path/PathTests/TestPathDrillable.py", line 272, in test20
    self.assertEqual(len(results), 20)
AssertionError: 21 != 20

----------------------------------------------------------------------
Ran 1098 tests in 56.051s

FAILED (failures=2)

```

luzpaz, 2024-12-23:
@coldtobi sorry for the drawn out process here. Is this still reproducible for you on v1.0 ?

coldtobi, 2024-12-23:
@luzpaz  not there yet, sorry... I've just uploaded version 1.0 to Debian, but as there is a package rename involved, this first needs to pass the [NEW](https://wiki.debian.org/NewQueue) queue and then I can investigate.

(There's still some work to be done, e.g the test suite is currently failing in the build environment (basically a chroot), but passes when built outside, so I need to dig into this furhter why that happens...)

sliptonic, 2025-10-04:
The tests pass on my machine.  The problem is specific to one architecture and I don't have the means to reproduce it.
If I had to guess, I would say the problem is one of tolerance.  If someone can reproduce it reliably, they could selectively start commenting out parts of test10 to find out exactly which assertion is failing.  Or better yet, break the test apart into individual tests.

FEA-eng, 2025-12-14:
I assume it's still relevant (see comments above)

coldtobi, 2026-03-22:
> ping [@coldtobi](https://github.com/coldtobi)

@luzpaz  I've just done an test upload of 1.0.2 to Debian experimental, let's see how it works out.

closed 2026-04-01 by sliptonic