FreeCAD/FreeCAD #7482: [Bug]  MacOS crash at startup all the time (critical)

krl91, 2022-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=71835&p=626089#p626089

### Version

0.20 (Release)

### Full version info

```shell
MacBook Pro:
2,9 GHz Intel Core i5
16 Go or RAM
Intel Iris Graphics 550 1536 Mo
```

### Issue description

I'm using MacOS 12.5.1 (MacIntel) and just installed last Downloadable version of FreeCAD for MacOS (20.1)
When I launch it I have the small window that is showing the loading of FreeCAD and the it crah all the time itself, so it sound to be a critical bug.... so I go back on FreeCAD 19 that works well..

Same issue on FreeCAD 0.21.
FreeCAD 0.19 works 

### Anything else?

`/Applications/FreeCAD.app/Contents/MacOS/FreeCAD --write-log
# packages in environment at /Users/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/osx/APP/FreeCAD.app/Contents/Resources:
#
# Name                    Version                   Build  Channel
aom                       3.4.0                hb486fe8_1    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
arpack                    3.7.0                hefb7bc6_2    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
blas                      2.116                  openblas    conda-forge
blas-devel                3.9.0           16_osx64_openblas    conda-forge
blinker                   1.4                        py_1    conda-forge
boost                     1.74.0          py310h509978a_5    conda-forge
boost-cpp                 1.74.0               h8b082ac_8    conda-forge
brotli                    1.0.9                h5eb16cf_7    conda-forge
brotli-bin                1.0.9                h5eb16cf_7    conda-forge
brotlipy                  0.7.0           py310h1961e1f_1004    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2022.6.15            h033912b_0    conda-forge
cairo                     1.16.0            h56d7d38_1012    conda-forge
calculix                  2.20                 h61d0cde_0    conda-forge
certifi                   2022.6.15       py310h2ec42d9_0    conda-forge
cffi                      1.15.1          py310h96bbf6e_0    conda-forge
charset-normalizer        2.1.0              pyhd8ed1ab_0    conda-forge
coin3d                    4.0.0             hf16527e_1011    conda-forge
cryptography              37.0.4          py310h52c3658_0    conda-forge
curl                      7.83.1               h372c54d_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
docutils                  0.19            py310h2ec42d9_0    conda-forge
double-conversion         3.2.0                he49afe7_0    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
expat                     2.4.8                h96cf925_0    conda-forge
ezdxf                     0.18            py310habb735a_0    conda-forge
ffmpeg                    4.4.2           gpl_h5a1d76f_107    conda-forge
flann                     1.9.1             h30321d8_1010    conda-forge
fltk                      1.3.8                h7ffa276_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
[... 202 lines cut ...]
xorg-xextproto            7.3.0             h35c211d_1002    conda-forge
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.6                h775f41a_0    conda-forge
yaml                      0.2.5                h0d85af4_2    conda-forge
zlib                      1.2.12               hfe4f2af_2    conda-forge
zstd                      1.5.2                hfa58983_4    conda-forge
FreeCAD 0.20.1, Libs: 0.20.1R29410 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

[0912/192441.802114:ERROR:icu_util.cc(251)] Couldn't mmap icu data file
/Applications/FreeCAD.app/Contents/MacOS/FreeCAD: line 19:  6079 Trace/BPT trap: 5       "${PREFIX}/bin/freecad" $@
`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pcontezini, 2022-09-16:
Working fine here, same version of Freecad running on Intel MacOS 12.6

It seems to be a problem with your QT library, check if you have it installed somewhere in your system and is loading from there instead of the embedded library (LD_LIBRARY_PATH?)

https://bugzilla.redhat.com/show_bug.cgi?id=1920379

rbuckland, 2022-09-18:
Also have this issue:

1. seems a few people have had the issue - https://forum.freecadweb.org/viewtopic.php?style=10&t=71737
2. am unable to login to freecadforum to report on that bug (registration not sending activation email) 

**OS Info**
- macOS Big Sur, 11.4
- 3.4Ghz , 8GB RAM
- 550G Free

**Tried**
- `rm -rf ~/Library/Preferences/FreeCAD ~/Library/Application\ Support/FreeCAD`
- Uninstall and Re-install

**debug log ..**
```
...>8...>8...>8...>8...>8...>8...>8...
FreeCAD 0.20.1, Libs: 0.20.1R29410 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

[0918/164453.258879:ERROR:icu_util.cc(251)] Couldn't mmap icu data file
/Applications/FreeCAD.app/Contents/MacOS/FreeCAD: line 19: 41930 Trace/BPT trap: 5       "${PREFIX}/bin/freecad" $@
Ramons-iMac:~ rbuckland$ rm -rf ~/Library/Preferences/FreeCAD ~/Library/Application\ Support/FreeCAD
Ramons-iMac:~ rbuckland$ /Applications/FreeCAD.app/Contents/MacOS/FreeCAD --write-log
# packages in environment at /Users/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/osx/APP/FreeCAD.app/Contents/Resources:
#
# Name                    Version                   Build  Channel
aom                       3.4.0                hb486fe8_1    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
arpack                    3.7.0                hefb7bc6_2    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
blas                      2.116                  openblas    conda-forge
blas-devel                3.9.0           16_osx64_openblas    conda-forge
blinker                   1.4                        py_1    conda-forge
boost                     1.74.0          py310h509978a_5    conda-forge
boost-cpp                 1.74.0               h8b082ac_8    conda-forge
brotli                    1.0.9                h5eb16cf_7    conda-forge
brotli-bin                1.0.9                h5eb16cf_7    conda-forge
brotlipy                  0.7.0           py310h1961e1f_1004    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2022.6.15            h033912b_0    conda-forge
cairo                     1.16.0            h56d7d38_1012    conda-forge
calculix                  2.20                 h61d0cde_0    conda-forge
certifi                   2022.6.15       py310h2ec42d9_0    conda-forge
cffi                      1.15.1          py310h96bbf6e_0    conda-forge
charset-normalizer        2.1.0              pyhd8ed1ab_0    conda-forge
coin3d                    4.0.0             hf16527e_1011    conda-forge
cryptography              37.0.4          py310h52c3658_0    conda-forge
curl                      7.83.1               h372c54d_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
docutils                  0.19            py310h2ec42d9_0    conda-forge
double-conversion         3.2.0                he49afe7_0    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
expat                     2.4.8                h96cf925_0    conda-forge
ezdxf                     0.18            py310habb735a_0    conda-forge
ffmpeg                    4.4.2           gpl_h5a1d76f_107    conda-forge
flann                     1.9.1             h30321d8_1010    conda-forge
fltk                      1.3.8                h7ffa276_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
[... 200 lines cut ...]
xorg-libxfixes            5.0.3             h0d85af4_1004    conda-forge
xorg-libxrender           0.9.10            h0d85af4_1003    conda-forge
xorg-renderproto          0.11.1            h0d85af4_1002    conda-forge
xorg-xextproto            7.3.0             h35c211d_1002    conda-forge
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.6                h775f41a_0    conda-forge
yaml                      0.2.5                h0d85af4_2    conda-forge
zlib                      1.2.12               hfe4f2af_2    conda-forge
zstd                      1.5.2                hfa58983_4    conda-forge
FreeCAD 0.20.1, Libs: 0.20.1R29410 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

[0918/164603.720834:ERROR:icu_util.cc(251)] Couldn't mmap icu data file
/Applications/FreeCAD.app/Contents/MacOS/FreeCAD: line 19: 42005 Trace/BPT trap: 5       "${PREFIX}/bin/freecad" $@
Ramons-iMac:~ rbuckland$ 
```

rbuckland, 2022-09-18:
This bug is related to compile parameters on qtwebengine it seems; 
https://github.com/Homebrew/homebrew-core/issues/104008#issuecomment-1159845166

Freecad is using 
```
qt-webengine              5.15.4               h72ca1e5_3    conda-forge
```

rbuckland, 2022-09-18:
Tested the lateste weekly-bundle and it has the same issue

    `FreeCAD_weekly-builds-30377-2022-09-15-conda-macOS-x86_64-py310.dmg`

```
[0918/204855.489244:ERROR:icu_util.cc(251)] Couldn't mmap icu data file
/Applications/FreeCAD.app/Contents/MacOS/FreeCAD: line 19: 47074 Trace/BPT trap: 5       "${PREFIX}/bin/freecad" $@
```

rbuckland, 2022-09-18:
digging a little deeper

1. the file is expected to be in `/Applications/FreeCAD.app/Contents/Resources` - https://lists.qt-project.org/pipermail/qtwebengine/2015-February/000157.html
  but it is found in `/Applications/FreeCAD.app/Contents/Resources/resources/`
  ```
  Ramons-iMac:~ rbuckland$ find /Applications/FreeCAD.app -name icudtl.dat 
  /Applications/FreeCAD.app/Contents/Resources/resources/icudtl.dat
  ```

The file that attempts to load the ICU file is Chromium - presumably qtwebengine is loading it
ref: https://chromium.googlesource.com/chromium/src/base/+/master/i18n/icu_util.cc#251
(note: i cant tell if this is the same version as bundled inside FreeCAD - but line number is close (offby-1 perhaps)

I have tried locating the file `icudtl.dat` into `/Applications/FreeCAD.app/Contents/Resources` but no joy - will keep digging.

mishka-zz, 2022-10-04:
I can confirm that symlinking the icudtl.dat and *.pak files into /Applications/FreeCAD.app/Contents/Resources/ will make it work.

leonelhs, 2022-10-16:
This works for me:

```
chdir /Applications/FreeCAD.app/Contents/Resources
ln -s resources Resources
```

luzpaz, 2022-10-17:
CC @ipatch

luzpaz, 2022-11-15:
Closed dupe #7787 in favor of this ticket. Can we get some traction on this ?

WelchITConsulting, 2022-11-16:
> This works for me:
> 
> ```
> chdir /Applications/FreeCAD.app/Contents/Resources
> ln -s resources Resources
> ```

Thanks this worked for me too.

luzpaz, 2022-11-16:
Hi @looooo, can we tweak the conda macos build with this symlink fix ?

looooo, 2022-11-19:
I hope this does the job:
https://github.com/FreeCAD/FreeCAD-Bundle/commit/2a602fbea9479434e84ab66f40bd601518ef9622

luzpaz, 2022-12-01:
Thank!
Can any macOS'ers check if this fixed it ?

pavelrevak, 2023-01-28:
Installing version from: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds, there is for MacOS - ARM version from 2022-12-31 and still is not fixed there, but fix with symlink helped me.

I tried only rename `FreeCAD.app/Contents/Resources/resources` to `FreeCAD.app/Contents/Resources/Resources/` and also it helped, so symlink is useless in that case, just use right folder name with upper case first letter.

On my MAC I always use use case-sensitive file system, so please do not forget on us

challengy1, 2023-03-19:
This information made me happy. Making a symbolic link worked well.
`ln -s resources Resources`
Three days ago, I reformatted my system disk for Ventura clean installation to make my Mac case sensitive. Installing FreeCAD 20.2, I encountered a crash on startup.
When I tried with 21.0 weekly build, the situation was same. 
I hope this issue may be resolved in next release. I don't think many Mac user switch to case sensitive but some will do.
Thank you everybody. Chao.

krl91, 2023-04-06:
Could someone correct this bug on the stable and official version?  so sad it takes years for a cast issue to be corrected....

luzpaz, 2023-04-06:
@chennes is this something you can correct in the macOS conda build ?