FreeCAD/FreeCAD #7486: [Bug] [MSVC] FreeCAD failed to build with error C2036 on MSVC

fangzhouxia, 2022-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=4&t=71889

### Version

0.21 (Development)

### Full version info

```shell
N/A
```

### Issue description

Hi all,
FreeCAD fails to build on MSVC due to error C2036. Could you help look?
**Error message**
C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.29.30133\include\vector(1946,1): error C2036: 'TechDraw::LineSet *': unknown size (compiling source file F:\gitP\FreeCAD\FreeCAD\src\Mod\TechDraw\App\AppTechDraw.cpp) [F:\gitP\FreeCAD\FreeCAD\build_amd64\src\Mod\TechDraw\App\TechDraw.vcxproj]
**Detailed log**
[build (5).log](https://github.com/FreeCAD/FreeCAD/files/9563230/build.5.log)
**Expected Behaviour**
build successfully.
**version info**
OS Version: windows server 2019
VS version: VS2019 (16.11.18)
the commit of FreeCAD we use is dcf2bc4

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2022-09-14:
Werner has already fixed this - https://forum.freecadweb.org/viewtopic.php?p=626194#p626194

closed 2022-09-14 by WandererFan

luzpaz, 2022-09-15:
Specifically 8b4e114b9d

fangzhouxia, 2022-09-16:
@WandererFan Hi, according to this - https://forum.freecadweb.org/viewtopic.php?p=626194#p626194 error C2036 has indeed been resolved. But a new problem appeared in the runtest stage, as follows. Do I need to re-issue this issue?
Traceback (most recent call last):
  File "F:\gitP\FreeCAD\FreeCAD\build_amd64\Mod\Part\TestPartApp.py", line 230, in testPlacement
    self.assertTrue(box.Placement.isSame(p2 * p1))
AssertionError: False is not true

----------------------------------------------------------------------
Ran 974 tests in 66.711s

FAILED (failures=1)
			(25 %)	
						(33 %)	
						(41 %)	
						(50 %)	
						(58 %)	
						(66 %)	
						(75 %)	
						(83 %)	
						(91 %)	
						(100 %)

WandererFan, 2022-09-16:
Doesn't look related to the other issue.  I would make a post about this in
the Install/Compile section of the forum.

fangzhouxia, 2022-09-20:
Hi, @ WandererFan could you give me a link to the forum, I can't find it

WandererFan, 2022-09-20:
https://www.forum.freecadweb.org/viewforum.php?f=4

luzpaz, 2022-09-20:
Per wwmayer in the [thread](https://forum.freecadweb.org/viewtopic.php?p=626194#p626194): 
> Your version is too old. The issue has been fixed with [git commit 8b4e114b9d](http://github.com/FreeCAD/FreeCAD/commit/8b4e114b9d)