FreeCAD/FreeCAD #7489: [Bug] Sketcher: Using Formula editor for Constrain input

balrobs, 2022-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=13&t=71781

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30345 (Git)
Build type: Release
Branch: master
Hash: f42a47aa0fd8c4f34ef63c185d9f29b9b6e84737
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3DfindIT.backup1663070439.7435825
  * Assembly3 0.11.3
  * BIM 2021.12.0
  * ExtremeProDark 1.0.4
  * FreeCAD_SketchArch
  * Help 1.0.3
  * ProDarkThemePreferencePack 1.0.0
```

### Issue description

Steps to reproduce:
1. unzip attached file and open it
2. double click on sketch "Haken" of "Pad001"
3. double click on  Constraint29 (24mm)
4. for Lenght open the Formula editor and input following expression: Pocket.Shape.Edges[5].Vertexes[0].Point.z
5. note the given result value: 21,21
6. exit the Formula editor dialog by clicking OK
7. note that Length value remains 24,00 mm
8. exit the Insert length dialog by clicking OK
9. note that the Constrain value remains 24,00 mm and is not updated to 21,21 mm

Additional information:
if after step 7 you insert an optional Name for the constraint (i.e. LL) then everytinbg works as expected.

[Haken und Zapfen V3 mit Tabelle.zip](https://github.com/FreeCAD/FreeCAD/files/9566749/Haken.und.Zapfen.V3.mit.Tabelle.zip)

### Anything else?

In Realthunder Linkstage3 the problem can't be seen.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2022-09-15:
I can narrow down the regression. The last build I have that the expression updates as expected is:

```
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.20.26155 (Git) 2021/10/16
Build type: Release
Branch: master
Hash: 0926a4148bcff11249fd4f56bc30256102ffe105
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)
```

The first build that it fails is:

```
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.20.26845 (Git) 2021/01/02
Build type: Release
Branch: master
Hash: 4430e3ec8c8050560478564a493e925bd7c5985a
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)

```

Syres916, 2022-09-15:
This does look very similar to #7390 When I get the commit for the regression, I'll test against the other example and confirm if it's a duplicate.

Syres916, 2022-09-15:
This has turned into a bit of a nightmare because I can't compile all the commits around the regression so the best I can state is the last working build info is:

Version: 0.20.26733 (Git) 22nd Dec 2021
Hash: 3d1b9f5c90a2578675494ed5f0967374756973e1

and the first failed build info is:

Version: 0.20.26738 (Git) 22nd Dec 2021
Hash: 7d23d5b9842a49cd54598c9c1f180f3a5184e079

The list of commits for that day is https://github.com/FreeCAD/FreeCAD/commits/master?after=860e50b589ea1d2083aa2945248903e14ba7ce1c+3604&branch=master&qualified_name=refs%2Fheads%2Fmaster

Syres916, 2022-09-15:
@realthunder I've triaged this bug as best as I can, if you have time could you please have a look at a possible fix, thanks in advance.

adrianinsaval, 2022-11-23:
is #7589 related or maybe even the same underlying bug?

luzpaz, 2022-11-29:
@balrobs and @Syres916 please respond at your convenience :point_up:

acpopescu, 2023-01-29:
I've hit this exact issue - see https://forum.freecadweb.org/viewtopic.php?f=10&t=75629&p=656725#p656725 

I think the issue is the buildGraph function in the PropertyExpressionEngine, it will sometimes SKIP valid properties to evaluate. Details are in the thread, plus a side-effect :( related to precision....

My local Fix branch : https://github.com/acpopescu/FreeCAD/tree/buildGraphFix.

tag @adrianinsaval, @luzpaz

luzpaz, 2023-02-22:
@acpopescu can this be closed ?

acpopescu, 2023-02-23:
Yes, it should be fixed by that merge.

closed 2023-02-23 by luzpaz

donovaly, 2023-03-27:
The fix will be available with the next release FreeCAD 0.20.3