FreeCAD/FreeCAD #7501: [Feature Request] improved quality of Thumbnail.png

MatthiasWM, 2022-09-15:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=10&t=71803#p626576

### Idea description

Thumbnails are saved by generating graphics at 128x128 (by default) without antialiasing (macOS Monterey, M1, HIDPI), making previews very boxy, especially when they are scaled when viewed.

I suggest to either enable antialiasing (which may not be cross-platform), or better, render the thumbnail at 4x resolution and then scale it down. This is cross-platform, the time to scale the image is minimal, and the image quality should be a lot better.

### Anything else?

I'll be happy to implement this when I have the time. A quick look at the source code points me at `GUI/Thumbnail.ccp`, line 98: `this->viewer->imageFromFramebuffer(this->size, this->size, 0, invalid, img);` .

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MatthiasWM, 2022-09-15:
Without oversampling:
[image]

With oversampling:
[image]

```
diff --git a/src/Gui/Thumbnail.cpp b/src/Gui/Thumbnail.cpp
index aa41e9686b..fd85e19d9c 100644
--- a/src/Gui/Thumbnail.cpp
+++ b/src/Gui/Thumbnail.cpp
@@ -95,7 +95,10 @@ void Thumbnail::SaveDocFile (Base::Writer &writer) const
         }
 
         QColor invalid;
-        this->viewer->imageFromFramebuffer(this->size, this->size, 0, invalid, img);
+        QImage scaledImg;^M
+        this->viewer->imageFromFramebuffer(this->size*4, this->size*4, 0, invalid, scaledImg);^M
+        if (!scaledImg.isNull())^M
+            img = scaledImg.scaled(this->size, this->size, Qt::IgnoreAspectRatio, Qt::SmoothTransformation);^M
     }
 
     // Get app icon and resize to half size to insert in topbottom position over the current view snapshot
@@ -105,7 +108,7 @@ void Thumbnail::SaveDocFile (Base::Writer &writer) const
         if (App::GetApplication().GetParameterGroupByPath
             ("User parameter:BaseApp/Preferences/Document")->GetBool("AddThumbnailLogo",true)) {
             // only scale app icon if an offscreen image could be created
-            appIcon =  appIcon.scaled(this->size / 4, this->size /4);
+            appIcon = appIcon.scaled(this->size / 4, this->size /4, Qt::IgnoreAspectRatio, Qt::SmoothTransformation);^M
             px = BitmapFactory().merge(QPixmap::fromImage(img), appIcon, BitmapFactoryInst::BottomRight);
         }
         else {
```

adrianinsaval, 2022-09-21:
if you want to propose a change in code it's better to make a pull request

closed 2022-09-28 by wwmayer