FreeCAD/FreeCAD #7510: transparency disapear by export obj and reimport

Hubhubhurra, 2022-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=72003

### Version

0.21 (Development)

### Full version info

```shell
OS: Debian GNU/Linux bookworm/sid (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.30398 (Git) AppImage
Build type: Release
Branch: master
Hash: b3dfdc568a92810449bf47bc73b6784d65dfa206
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Arch

### Issue description

If I set the transparency to some value and export this to obj and reimport the file the transparency disappeared.
Hint:
In the .mtl-file exist for each color a line
Tr ...
with the right value. But this is not the modern way to write it and by import FreeCAD read instead rightly the
d ...
line too. But in false direction, then d is complementary to Tr and by import the d-value FreeCAD use it as Tr-value (instaed of 1-d).
Solution:
In Mod/Arch/importOBJ.py (and also src/Mod/Arch/importOBJ.py)
line 254 and 261 I changed "Tr " to "d "
and recompile. As result import and export by FreeCAD self ar good, but not in conjunction with other programms. So I must change line 261 to
outfile.write("d " + str(1-mat[2]/100) + "\n")
and other programs (i. e. Sweet Home 3D) can import the files in right way, but now FreeCAD not. To correct this I has change line 344 to
trans = int((1-float(mline[2:]))*100)
Than all works fine.

On attention to programs using the Tr line it can export this line too leave
outfile.write("Tr " + str(mat[2]/100) + "\n")
and add after that
outfile.write("d " + str(1-mat[2]/100) + "\n")

### Anything else?

[demofiles.zip](https://github.com/FreeCAD/FreeCAD/files/9604209/demofiles.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-09-21:
Thanks, Can you make a PR ?

luzpaz, 2022-10-15:
@yorikvanhavre can this ticket be closed ?

Hubhubhurra, 2022-10-16:
Yes, now it works fine.

closed 2022-10-16 by Hubhubhurra