FreeCAD/FreeCAD #7516: [Bug] <UI consistency of Tasks breaks with the Appearance & Measurement tool's task UI>

rahmanshaber, 2022-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=623293#p623293

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.0
  * kicadStepUpMod 10.14.0
[/code]
```

### Issue description

All the tool's task UI's "Cancel" and "OK" button in top except the measurement tool's. Same is true for appearance tool's task.
[image]
[image]

These tool's task page have the buttons in top
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Zolko-123, 2022-10-28:
This is not a [Bug] and the title should be renamed.

luzpaz, 2022-10-28:
How to characterize it then, suggestions?

luzpaz, 2022-10-29:
It is discussed on the forum. It is being tracked here so we don't forget. It's also labeled UI/UX

adrianinsaval, 2022-11-04:
Buttons are now on top so that part of the issue is tackled, should this be kept open for the cancel buttons? IMO a cancel button doesn't make sense for the measurements tool
the appearance dialog would need some refactoring to get a cancel button I think

chennes, 2022-11-04:
I think that adding Cancel to appearance is a significant enough amount of work to warrant it's own issue -- my preference is to close this one and create a new, focused issue, and cross-link them.

luzpaz, 2022-11-04:
@adrianinsaval thank you!

@chennes I agree. Lets close and open an issue for 'Cancel' button.

rahmanshaber, 2022-11-04:
@adrianinsaval that cancel button in the measurement tool won't put the measured value after you close that dialog. But if user clicks okay then the measurements will be there with the part

adrianinsaval, 2024-07-21:
this measurement tool has been removed so this is no longer relevant

closed 2024-07-21 by adrianinsaval