FreeCAD/FreeCAD #7517: [Feature Request] Add cancel button in the Transform task UI

rahmanshaber, 2022-09-22:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=34&t=71598

### Idea description

Sometime we just move a part to compare with another, but to go back to previous position we need to click OK and then undo the changes.
But if the part is filled with details then it takes time to undo the changes. So with a "Cancel" button this can be prevented.
Also if you move the part in 6 step than you have to press ctrl+z 6 times to go to previous position.

With the cancel button that will be reduced and the UI consistency will also be maintained. 

[image]

Requested UI 
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-11-08:
@adrianinsaval do you want to tackle this? :stuck_out_tongue_winking_eye:

adrianinsaval, 2022-11-08:
I agree with the request but don't really know how to do it so it's gonna be a pass from me for the time being, perhaps at a later time I'll learn some about all the transaction and undo mumbo jumbo.

luzpaz, 2023-06-11:
Per wwmayer in https://github.com/FreeCAD/FreeCAD/pull/9647#issuecomment-1560690244

> Just by adding a Cancel button the issue is not solved because when clicking the button after a transformation the geometry doesn't jump back.
> 
> What's still needed is:
> 
>     * the transaction handling must be moved from ViewProviderDragger (dragStartCallback/dragFinishCallback) to TaskCSysDragger  
>     * TaskCSysDragger::accept() must commit the pending transaction
>     * implement TaskCSysDragger::reject() that just aborts a pending transaction

Aryan01b, 2023-08-19:
@luzpaz, Can i work on this issue.
if so, then Which file should be modified to add the feature?

luzpaz, 2023-08-21:
@Aryan01b perhaps @PaddleStroke can give some insight ?

PaddleStroke, 2023-08-21:
Same as Adrian here, I don't know the transaction system.
But wwmayer has already said what needs to be done in the message you
quoted no?

luzpaz, 2023-08-21:
Thanks for responding @PaddleStroke, just for clarification, I was quoting wwmayer from a closed PR (not a direct message). It's not clear what file(s) he was referring to specifically.

didendron, 2023-08-31:
@luzpaz, Could I pick this as my first issue?

luzpaz, 2023-09-01:
@didendron go for it!

PR #10529 (2023-09-11): Gui: Add cancel button to transform task UI dialog
Fixes #7517

This PR replaces https://github.com/FreeCAD/FreeCAD/pull/10518 because something with the merging went wrong.
commit 23512195 (2023-09-12): Gui: Add cancel button to transform task UI dialog

Fixes #7517
commit 7e0c640c (2023-11-10): Gui: Add cancel button to transform task UI dialog

Fixes #7517