FreeCAD/FreeCAD #7520: [Bug] Angle Constraint: Line-Point-Line Mode - auxiliary arc lines and value positioning missing

PLChris, 2022-09-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=70325

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.30398 (Git)
Build type: Release
Branch: master
Hash: b3dfdc568a92810449bf47bc73b6784d65dfa206
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * FeedsAndSpeeds 0.4.0
  * fcgear 1.0.0
  * ShortCuts
  * fasteners 0.4.6
  * sheetmetal 0.2.57
  * ExplodedAssembly
  * Curves 0.5.8
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

The special angle constraint mode selecting two lines and a point doesn't show he arc of the angle and position of the numeric value is off.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Zolko-123, 2022-10-28:
Not enough information, should be closed. (what would be the expected behavior ?)

PLChris, 2022-10-29:
> Not enough information, should be closed. (what would be the expected behavior ?). And it's certainly not a [Bug] at this stage

It is all in the linked forum post. It used to work in the same way as the other angle constraints, so I still rate it as a bug.

luzpaz, 2022-11-07:
@PaddleStroke if you have a moment, can you take a look at this ?

PaddleStroke, 2022-11-08:
I had a quick look, but couldn't find the root. The drawing is a part of the code that I don't know so it's hard to find.

AjinkyaDahale, 2023-04-14:
There seems to be an issue computing normals. I'm having similar issues with #9125 where the constraint is applied correctly but the lines are wrong. From what I can tell, the error is coming in `GeomCurve::normalAt`, where the catch statement is triggered.

https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Part/App/Geometry.cpp#L665-L688

Disclaimer: this inference is from unmerged code, and the exact mechanism of the bug here may be different.

PR #9681 (2023-05-28): Sketcher: Fix bug angle via point visualisation
===============================================

Issue description:
https://forum.freecad.org/viewtopic.php?f=3&t=70325 https://github.com/FreeCAD/FreeCAD/issues/7520

The general algorithm to calculate normals miserably fails when the curve is a line segment and the point through which the normal to a curve is to be drawn is colinear with the line segment.

The solution is to provide specific code for this corner case.

fixes #7520
commit 1d61ed18 (2023-05-28): Sketcher: Fix bug angle via point visualisation

===============================================

Issue description:
https://forum.freecad.org/viewtopic.php?f=3&t=70325
https://github.com/FreeCAD/FreeCAD/issues/7520

The general algorithm to calculate normals miserably fails when
the curve is a line segment and the point through which the normal
to a curve is to be drawn is colinear with the line segment.

The solution is to provide specific code for this corner case.

fixes #7520
closed 2023-05-28 by abdullahtahiriyo