FreeCAD/FreeCAD #7530: Image open and import fails

gicking, 2022-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

Cannot import pictures

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29177 +233 (Git) Snap 245
Build type: Release
Branch: tag: 0.20.1
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]
```

### Issue description

Hello,

first of all thanks a lot for developing and maintaining FreeCAD! Great job :-)

Unfortunately I have a problem importing images in the Image workbench. According to [this](https://www.youtube.com/watch?v=3caDRzc87_Y) or [this](https://www.youtube.com/watch?v=xorYJELqTxw) tutorial video, I should be able to view or import pictures from the Image workbench without further preparation. But when I press the respective buttons, it has no effect. Namely, the commands show up in the Python console (see screenshot), but no file selection box opens. What am I missing? 

I am using Xubuntu 20.04.5 (amd64) and FreeCAD 0.20.1 installed via system Snap-Store. So nothing funny there. If you need more information, just let me know. 

Thanks again for your support and have a great day!

Georg

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gicking, 2022-09-27:
Update: 

I deinstalled the snap and 

- installed via apt-get
  - the installed version is 0.18.4 -> outdated
  - image view and import works as expected
  - self-tests fail altogether, because some library is not found
  - deinstalled again

- downloaded the AppImage from [https://www.freecadweb.org](https://www.freecadweb.org)
  - installed version is 0.20.1.29410
  - image view and import works as expected
  - self-tests result in only 1 error (see attachment)
[image]

So either there were many bugfixes (including image import) between 0.20.1.29177 (Snap) and 0.20.1.29410 (AppImage), or there is an issue with the Snap package. I suspect the latter.

For now I will stick to the AppImage as it solves the initial problem.

luzpaz, 2022-10-03:
In the future, you need to sign-up to the forum and create an issue. It saves you and everyone work primarily because development versions are way ahead of minor versions. Can you confirm with Snap 'edge' ? 

```shell
snap refresh freecad --channel=latest/edge # switch to edge/nightly version
snap refresh freecad --channel=latest/stable # switch back to stable version
```