FreeCAD/FreeCAD #7532: Scroll changes numerical values in property editor

Ognian, 2022-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?style=3&t=30772

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: C/Default (C)
Installed mods: 
  * FreeCAD_SketchArch
  * BIM 2021.12.0
  * Curves 0.5.8
[/code]
```

### Issue description

Although marked as fixed this still happens on macOS Monterey 12.6
This is a big Problem since it is also not possible to get the previous values…

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-09-27:
have you checked in the development version ?
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Ognian, 2022-09-27:
Tested with:
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.30454 (Git)
Build type: Release
Branch: master
Hash: 5df954690787dedbbf3c35f658f010f388474676
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FreeCAD_SketchArch
  * BIM 2021.12.0
  * Curves 0.5.8
[/code]
witch is 2 days old
unfortunately the behavior is the same...

Ognian, 2022-09-28:
Using
Parameter Editor >BaseApp >Preferences >General >ComboBoxWheelEventFilter =true
Doesn‘t help eigther

Ognian, 2022-10-04:
@luzpaz ComboBoxWheelEventFilter=true doesn't seem to work anymore
on a mac this is pretty frustrating...
tested with the latest without any additional mods
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.30486 (Git)
Build type: Release
Branch: master
Hash: eb546e25d3f952869e3ec87dca02b58653cb3936
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
[/code]
Is this due to the latest changes in Qt?

luzpaz, 2022-10-05:
Please participate on the forum thread. I 're-opened' the thread and bumped it requesting macOS testers

luzpaz, 2023-04-15:
@chennes was this solved with ccbc9612e  
@Ognian can you test macOS development build to see if this is still an issue?

0penBrain, 2023-04-16:
Yes, should be solved

luzpaz, 2023-04-16:
Thx!  
Closing. If issue persists, please re-open.

closed 2023-04-16 by luzpaz