FreeCAD/FreeCAD #7546: [Feature Request] Concentric constraint in the Sketcher

FEA-eng, 2022-10-03:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=72241

### Subproject(s) affected?

Sketcher

### Idea description

Commercial CAD software typically features concentric constraint for sketches. It's similar to coincident constraint applied to circle/arc centers but has some advantages over it (listed in the referenced thread).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Zolko-123, 2022-10-03:
> Commercial CAD software ...

this is a red flag: why should this be any argument ?

> It's similar to coincident constraint applied to circle/arc centers

no, it's **EXACTLY** the same per definition: con-centric means coincident centers. This issue should be closed as _solved_ and not burden already limited developers time.

FEA-eng, 2022-10-03:
> this is a red flag: why should this be any argument ?

Don't exaggerate, I know that FreeCAD veterans don't like comparisons with commercial software and that FreeCAD will never compete with that software but sometimes it's good to take a look at current standards in the CAD industry. After all, they were established by experts following the needs of users and are meant to make the designers' lives much easier.

> no, it's **EXACTLY** the same per definition: con-centric means coincident centers. 

Even if the concentric constraint internally works on centers like the coincident constraint, there's still a difference in the way they are used - concentric constraint is applied to arcs/circles themselves and coincident constraint is applied to their centers. The implications of this difference (and thus the advantages of concentric constraint) are listed in the referenced thread. Also, from what I know, concentricity and coincidence are not the same thing when 3D space is considered (this is also explained in that forum thread but is not relevant for the Sketcher since it can't handle 3D geometry - I just want to clarify that they are not necessarily the same by definition). And finally, whether you like this argument or not, all the commercial CAD programs offer both types of constraints and they do it for a reason.

> This issue should be closed as _solved_ and not burden already limited developers time.

Marking this FR as solved would be misleading for the reasons mentioned above. As a trade-off, it could be changed to something like "[FR] Possibility to apply the coincident constraint to arcs and circles themselves" since this would bring the advantages of concentric constraint to the Sketcher and would be sufficient at least until 3D sketches are introduced.

AjinkyaDahale, 2022-10-04:
> > Commercial CAD software ...
> 
> this is a red flag: why should this be any argument ?

Let's not get militant about this. Though I agree that the report should have begun with the advantages and mentioned that this exists elsewhere only in passing.

> > It's similar to coincident constraint applied to circle/arc centers
> 
> no, it's **EXACTLY** the same per definition: con-centric means coincident centers. This issue should be closed as _solved_ and not burden already limited developers time.

There are some clear implementation-based differences. There are a few advantages to having this separate method, for example when the radius is extremely large but the frame is small (like 1000mm radii arcs in a 20x20 frame).

The fact that developer time is stretched thin doesn't mean that we start closing lower priority issues. This one, for example, is a very good example of "low hanging fruit".

silopolis, 2022-10-04:
Le mar. 4 oct. 2022 à 10:03, Ajinkya Dahale ***@***.***> a
écrit :

> There are some clear implementation-based differences. There are a few
> advantages to having this separate method, for example when the radius is
> extremely large but the frame is small (like 1000mm [...]
Let me second that !

chennes, 2022-11-04:
Where do you envision this in the user interface? We've been working recently to trim up the number of buttons showing on the toolbar -- in order to display this constraint, we either have to sacrifice an additional toolbar slot for it, or combine it with another tool. Any suggestions for the best approach?

AjinkyaDahale, 2022-11-04:
> ...we either have to sacrifice an additional toolbar slot for it, or combine it with another tool. Any suggestions for the best approach?

@chennes with @PaddleStroke's recently merged PR on element widgets we have an alternative, which will also work if, say for whatever reason, we want to apply coincidence to one endpoint of one curve and center of another. Yet another alternative: if a circle is selecter, replace the coincident icon with a "concentric" one.

PaddleStroke, 2022-11-04:
My take on the matter: 
-> Just enable selection of 2 circles with coincidence tool. When user select 2 circles edges with coincidence tools, it creates a coincidence of their centers.
 This way coincidence tool acts as 'concentric tool' and we don't have more tools cluttering the constraint toolbar (which is already so crowded).

FEA-eng, 2022-11-04:
> Just enable selection of 2 circles with coincidence tool.

That's right, in the referenced forum thread we agreed that this should suffice for now.

PaddleStroke, 2022-11-04:
I have done this. It works as expected.
https://github.com/FreeCAD/FreeCAD/pull/7703
I will let you @FEA-eng follow the PR, find a merger willing to look into it and get it merged.

FEA-eng, 2022-11-04:
> I have done this. It works as expected. #7703 I will let you @FEA-eng follow the PR, find a merger willing to look into it and get it merged.

Great, thank you. I will add a link to this PR in the referenced forum thread and try to attract some mergers.

adrianinsaval, 2022-11-04:
Great job paddle, I think your solution is exactly what I had imagined for this

chennes, 2022-11-04:
It's been merged, thanks @PaddleStroke.

closed 2022-11-04 by chennes