FreeCAD/FreeCAD #7550: [Bug] [macos] build mac app bundle with homebrew deps will fail due invalid CMakeLists.txt

ipatch, 2022-10-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=4&t=72325

### Version

0.21 (Development)

### Full version info

```shell
not really applicable as this a build time issue, and not a run time.

**NA**
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

build time issue as described above. build time issue related to mac app bundling using homebrew deps.

### Anything else?

steps to reproduce.

the cmake configuration steps i use are,

```
cmake \
-DHOMEBREW_PREFIX=$bp \
-DPYTHON_EXECUTABLE="$bp/opt/python@3.10/bin/python3" \
-DPYTHON_INCLUDE_DIR="$bp/opt/python@3.10/Frameworks/Python.framework/Headers" \
-DCMAKE_BUILD_TYPE=RelWithDebInfo \
-DBUILD_FEM_NETGEN=0 \
-DFREECAD_USE_3DCONNEXION=0 \
-DBUILD_QT5=1 \
-DBUILD_ENABLE_CXX_STD=C++17 \
-Wno-dev \
-DFREECAD_CREATE_MAC_APP=1 \
-DCMAKE_INSTALL_PREFIX="$code/freecad-git/installs/$(basename $PWD)" \
-DCMAKE_PREFIX_PATH="$pthvtk;$pthqt;" \
../../freecad-src/

```

this issue is valid for both intel and m1 builds of freecad (should not make a difference)

the cmake argument in question is,

```
-DFREECAD_CREATE_MAC_APP=1 \
```

```
make install
```

error output, STDOUT, this is a snipped of the error from running the above commands,

```
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/lib/qtplugins/webview/libqtwebview_darwin.dylib
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/lib/qtplugins/webview/libqtwebview_webengine.dylib
-- Making bundle relocatable...
-- INFO: Analyzing bundle dependencies...
-- ERROR: Unable to find LC_DYLD_LOAD entry: @loader_path/libicudata.71.dylib
-- ERROR: Failed to resolve dependency in /Users/chris/homebrew/opt/icu4c/lib/libicuuc.71.dylib
Traceback (most recent call last):
  File "/Users/chris/code/freecad-git/freecad-src/src/Tools/MakeMacBundleRelocatable.py", line 391, in <module>
    main()
  File "/Users/chris/code/freecad-git/freecad-src/src/Tools/MakeMacBundleRelocatable.py", line 374, in main
    build_deps_graph(graph, bundle_path, dir_filter, search_paths)
  File "/Users/chris/code/freecad-git/freecad-src/src/Tools/MakeMacBundleRelocatable.py", line 241, in build_deps_graph
    deps = create_dep_nodes(list_install_names(k2), s_paths)
  File "/Users/chris/code/freecad-git/freecad-src/src/Tools/MakeMacBundleRelocatable.py", line 166, in create_dep_nodes
    raise LibraryNotFound(lib_name + " not found in given search paths")
__main__.LibraryNotFound: libicudata.71.dylib not found in given search paths
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/share/License.txt
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/share/examples/Schenkel.stp
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/share/examples/draft_test_objects.FCStd
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/share/examples/EngineBlock.FCStd
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/share/examples/PartDesignExample.FCStd
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/share/examples/RobotExample.FCStd
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/share/examples/ArchDetail.FCStd
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/share/examples/FemCalculixCantilever2D.FCStd
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/share/examples/FemCalculixCantilever3D.FCStd
-- Installing: /Users/chris/code/freecad-git/installs/build.git.master.001/FreeCAD.app/Contents/share/examples/FemCalculixCantilever3D_newSolver.FCStd
╭─chris ssh Chriss-Mac-mini in /Users/chris/code/freecad-git/builds/build.git.master.001
╰─λ     
```

basically the globbing patter for finding icu4c library needs to be updated. i can submit a PR in a bit. this is an issue i have addressed locally many times. just never bothered to report an issue.                                                                                               

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-10-08:
@ipatch this can be closed, right?

ipatch, 2022-10-08:
> @ipatch this can be closed, right?

yup.

closed 2022-10-15 by luzpaz