FreeCAD/FreeCAD #7551: [Feature Request] Support to create a compound from a dxf file

wwmayer, 2022-10-03:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=72126

### Subproject(s) affected?

File formats

### Idea description

The current dxf importer takes a long time to load bigger files (e.g. 15 MB) if too many entities are inside. The solution is to create a single compound object that holds all shapes of the file.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-15:
@kpemartin Any interest in tackling this DXF feature request ?

kpemartin, 2023-03-16:
I'm not sure I know enough about FreeCAD to do this. I'm not familiar with "compound objects".

Also, right now my driving goal is to be able to import the two or three dozen AutoCAD drawings I have with as little loss of fidelity as possible so I can finally shut off the old computer I'm keeping alive to run the old AutoCAD version I have. There are still at least half a dozen issues that must be tackled to get to this point. Some of these are pretty clearly bugs, and some of them are things impossible to represent in FreeCAD, but some of them might be deliberate decisions made in the past that can't just be changed because of a need to preserved backwards compatibility. Changes in the latter instance will be harder to get approved because of this.

As such anything that takes the resulting FreeCAD drawing further from the AutoCAD drawing is anathema to me, particularly when it is something that could be done with a post-processor on the finished imported drawing.

In this particular case, the work could be done by a post-processor (which would then be useful for any very complex drawing, not just one imported from AutoCAD), but it is difficult to tell if this would satisfy the OP. It is not clear whether their post relates to the time required to do the import, or also the time required to manipulate the imported drawing on an ongoing basis. If they import any particular drawing only once, and then use the FreeCAD drawing from then on, then I expect the latter is their main concern, and a post-processor would probably be sufficient for them.

It could also just be that the existing import code could be sped up. However, the OP does not specify whether they are using the legacy importer or the C++ one.

yorikvanhavre, 2023-03-20:
Note that ATM if the "group objects into layers" option is enabled, we already get more or less that result.
Maybe that option should be expanded into some drop-down thing like:

* each object as is
* one object by layer
* one object by file

It would be rather easy to implement, the code to "gather" objects into a compound is there aready, it would be just a matter of extending. I'm willing to tackle this, unless you want to, @kpemartin ;)

kpemartin, 2023-03-20:
Are you referring to "Group layers into blocks" perhaps? I don't see any "group objects into layers". Regarding "Group layers into blocks", what does that really mean anyway? FreeCAD doesn't seem to have any concept called "blocks" or at least a quick search did not reveal any such concept.

Again, my unfamiliarity with parts of FreeCAD may be confusing me. Are we referring here to making the same sort of compound object that the Part → Compound → Make Compound entry in the Part menu creates?

The original issue was about the time taken to import a large file, and I'm not sure that turning the entire drawing into a single object or each layer into a single object would have much effect on the import speed.

Doing so would very likely speed up manipulating the imported drawing since the entire layer/drawing would be a single entity in the underlying topology package, but I'm not sure it would speed up the import itself. One would do almost as well to have a separate post-processing step (one the user invokes separately) that does the merging. This would also give the ability (with a UI for this command) to be more selective about what is merged and would also work for drawings imported from other sources besides DXF/DWG.

In any case if you want to do that, you could use a combination of the existing "Group layers into blocks" and "Use Layers" options:
don't "Group layers into blocks" and don't "Use Layers": Individual drawing objects as direct children of the drawing
don't "Group layers into blocks" and "Use Layers": Individual drawing objects grouped into layers as they were in the DXF file
"Group layers into blocks" and don't "Use Layers": All drawing objects grouped into a single compound
"Group layers into blocks" and "Use Layers": The contents of each DXF layer grouped into a single compound each

Essentially this would interpret "Group layers into blocks" (which I feel is poorly worded for what it does anyway) as "Combine sibling objects into blocks"

I just did a little experiment and found that although I can place text in a Compound object if I make it using FreeCAD commands, neither importer places text in the Compound if "Group layers into blocks" is selected.

kpemartin, 2023-12-05:
I'll look into this one once I get #8806 done. Regarding text and dimensions, I guess they'll just stay as individual objects. If that is too much of a burden the user can choose to not import them at all.

kpemartin, 2024-01-01:
I may take a look at this one next. There is swirling interaction between "Use colors", "use layers" and "group objects" that I'd want to address. In particular if using colours from the DXF file only objects of the same color would be merged. This would still substantially decrease the number of objects, but if you want everything to merge you would have to skip importing the colours.

yorikvanhavre, 2024-01-08:
#11873

closed 2025-07-28 by yorikvanhavre