FreeCAD/FreeCAD #7557: [Feature Request] dimensioning diameters on flat lines

Jee-Bee, 2022-10-04:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=70867

### Subproject(s) affected?

Sketcher

### Idea description

**Assume that the 'revolve' method is used after wards**

In Engineering it is usual to design or recreate a object that has an cylindrical shape. In a lot of cases it is more natural do design (and for sure measure) diameters. 
What is the situation in this case: The sketch only have flat lines so all dimensions are basically _radii_.

**What is my idea:**  
When dimensioning from / to one of the axis it is possible to create a dimension that gives( and show) double the distance as the distance between the axis and the line. As workflow is see some options:
- tree clicks (line axis line)
- position of mouse cursor (other end of the axis is diameter)
- ...
For visual example see topic

_Note: Maybe a dedicated center line / axis is helpful so that non horizontal / vertical axis can be created. 
First i thought is comparable with the point, but the point currently don't have a function outside sketcher and this one should be select-able by the revolve..._

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jee-Bee, 2022-10-04:
Oke you can close this one i'm later than #7334

adrianinsaval, 2022-10-05:
as the person who opened the issue you can close it yourself @Jee-Bee

luzpaz, 2022-10-12:
Duplicate

closed 2022-10-12 by luzpaz