FreeCAD/FreeCAD #7573: [Feature Request] [Draft] Allow to turn dimension arrows on/off individually

yorikvanhavre, 2022-10-07:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=631252#p631252

### Subproject(s) affected?

Draft

### Idea description

Currently dimensions have a "ShowLine" property that switches on/off the display of both line and the arrows. Maybe we could separate this into 3 properties: 

* ShowLine: turns lines on/off
* ShowStartArrow: turns start arrow on/off
* ShowEndArrow: turns end arrow on/off

When loading an object that has no ShowStartArrow and ShowEndArrow properties, old behaviour should still apply (ShowLine controls arrows too)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

NXTler, 2023-08-26:
I like this, it would be very useful.

luzpaz, 2023-09-14:
Can we close ticket ?

Roy-043, 2023-09-16:
IMO we should leave it open. Draft Dimensions still require some work, and it would be a good idea for Future developers to be aware of this feature request.

luzpaz, 2023-09-16:
Can we delineate what is left to do  ?

yorikvanhavre, 2023-09-18:
Basically what is in the initial post stays valid here:

Currently the [Dimension view provider](https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/draftviewproviders/view_dimension.py) already has a "ShowLine" property which is [on by default](https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/draftviewproviders/view_dimension.py#L263). When off, the lines and arrows are [not drawn](https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/draftviewproviders/view_dimension.py#L709). This is done by switching subnode of a [SoSwitch](https://www.openinventor.com/reference-manuals/NewRefMan1030/RefManJava/com/openinventor/inventor/nodes/SoSwitch.html) on and off.

What needs to be done IMHO:

1. Instead of having the line and the arrows [inside a same SoSwitch](https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/draftviewproviders/view_dimension.py#L333) node, add three more SoSwitch inside that one (I would leave the current one in place, so we still have a "master" switch, that could still be useful) one for the line and one for each arrow.
2. Add Two more properties: ShowStartArrow and ShowEndArrow
3. Make the appropriate switch changes in the onChanged() method

Roy-043, 2023-10-07:
Alternatively we could go for two ArrowType properties: ArrowTypeStart and ArrowTypeEnd. If a "None" options is included in the enumeration, arrows can then be switched off individually as well.

yorikvanhavre, 2023-10-09:
Yes that's indeed far more elegant

CalligaroV, 2024-01-09:
Hi Everyone! :smiley:

After quite some time I've been finally able to use some of my time to go on with this feature request.
I just opened a PR with a first commit containing the modifications needed to meet @Roy-043's and @yorikvanhavre's suggestions.

I'm almost sure that something should be coded better, for example how to manage the "None" option, but at least this first commit is a starting point and during my tests the modifications seem working correctly.
Moreover, the functionality introduced by #10655 is still working.

Therefor I ask if someone of your can try it and give me a feedback, possibly in the PR conversation.

TIA!

PR #11941 (2025-08-14): [Feature Request] [Draft] [UI/UX] Allow to turn dimension arrows on/off individually
Fixes #7573

This PR adds the properties `ArrowTypeStart`, `ArrowTypeEnd` and all the code/settings/ui needed to manage these new two properties.
closed 2025-08-14 by kadet1090