FreeCAD/FreeCAD #7575: Deleting Rows in Spreadsheet WB sometimes crashes and causes unexpected results below deleted rows[Bug] <title>

PAS-DRAFTER, 2022-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=72403&sid=f7381c2d0822425001b4db671284bc75

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
  * Curves 0.5.6
  * ExplodedAssembly
  * fasteners 0.4.2
  * Manipulator 1.4.9
  * MOOC 2022.4.21
  * OpticsWorkbench 1.0.8
  * sheetmetal 0.2.56
  * ToolViewReverse_ToolBar
```

### Subproject(s) affected?

Spreadsheet

### Issue description

Removing a row in the spreadsheet causes merged cells below it to lose the proper formatting and alignment.  Removing multiple times can cause the program to crash.  The Forum link has a detailed description and a sample file that allows for the problem to be duplicated.  The problem was confirmed by two other users.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2022-11-04:
@PAS-DRAFTER I believe that this has been fixed by #7609 -- can you confirm?

PAS-DRAFTER, 2022-11-04:
@chennes Gladly.  Please excuse my ignorance, but how do I know what build has that fix so I know it is in the one I have loaded?  I'm new to this bug tracking stuff.

I currently have this on one machine.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30716 (Git)
Build type: Release
Branch: master
Hash: cbd19631428f1d9eb4faae30220551608a648589
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.0
  * CurvedShapes 1.0.3
  * Curves 0.4.4
  * DynamicData 2.46.0
  * fasteners 0.3.45
  * fcgear 1.0.0
  * MOOC 2022.4.21
  * sheetmetal 0.2.49
[/code]

PAS-DRAFTER, 2022-11-04:
@chennes I went out and downloaded the latest weekly build and installed on my other computer.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30716 (Git)
Build type: Release
Branch: master
Hash: cbd19631428f1d9eb4faae30220551608a648589
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
  * Curves 0.5.6
  * ExplodedAssembly
  * fasteners 0.4.2
  * Manipulator 1.4.9
  * MOOC 2022.4.21
  * OpticsWorkbench 1.0.8
  * 
  * sheetmetal 0.2.56
  * ToolViewReverse_ToolBar
[/code]

I can confirm that adding and removing rows no longer has an effect on the merged cells and I did it a lot to try and crash it, but I was not able to do that.

HOWEVER, further testing found that if I just deleted Blank row 7, and then hit control Z to undo what I just did, I find that cells B65, B66, B67, and a bunch more have turned red indicating there is an error in the formula.  Since I only deleted a blank row that was not referenced by anything else, I am not sure why that should have had an effect any of the calculations.

The Refresh button was active, so I pressed it, but the console just kicks out a bunch of errors and nothing changes.

If I close and save the file after this,  I can reopen the file and the Spreadsheet still shows that there are some errors in the tree,   I can then open the spreadsheet and hit Refresh and all the errors go away.

If I go and delete another row and undo, I get the same problem.

I also get the same problem if I Undo from the File Menu.

So, it looks like you guys fixed one issue and then another one has crept in.  Let me know if you need me to test anything else.

chennes, 2022-11-04:
I believe what you are referring to is https://github.com/FreeCAD/FreeCAD/issues/7511 -- can you confirm?

PAS-DRAFTER, 2022-11-04:
@chennes I would say the bug is similar to the one described there with a few exceptions.

1) I don't have to delete the top row.  I can delete other rows beside the top one and can get the system to show errors on the calculations. and the console (after a clear) shows this (There are a lot of calculations on this page, so there are a lot of errros);

`13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.D104: Property 'NozMntPltThk' not found in 'NozMntPltThk'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.B64: Property 'NozMntBar1Hgt' not found in 'NozMntBar1Hgt'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.D125: Property 'NozArmHor3' not found in 'NozArmHor3'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.B115: Property 'NozArmHor3' not found in 'NozArmHor3'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.B109: Property 'NozArmHor3' not found in 'NozArmHor3'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.B119: Cannot convert str to Quantity
in expression: NozMntBar3X - NozHole3SideOS
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.B117: Property 'NozMntBar1X' not found in 'NozMntBar1X'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.I110: Property 'NozArmLength3' not found in 'NozArmLength3'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.H74: Property 'NozArmLength3' not found in 'NozArmLength3'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.B66: Property 'BermNoz3Ang' not found in 'BermNoz3Ang'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.J109: Property 'NozArmLength3' not found in 'NozArmLength3'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.J105: Property 'Ext1' not found in 'Ext1'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.H113: Property 'NozMntBar1Wth' not found in 'NozMntBar1Wth'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.J113: Cannot convert str to Quantity
in expression: NozMntPltCut1Wth + 2 * NotchNarrowWth
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.I104: Property 'NozMntPltHeight' not found in 'NozMntPltHeight'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.B65: Property 'BermNoz2Ang' not found in 'BermNoz2Ang'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.J107: Property 'Ext2' not found in 'Ext2'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.H104: Property 'NozMntPltWidth' not found in 'NozMntPltWidth'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.E74: Property 'NozArmVer3' not found in 'NozArmVer3'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.C114: Property 'VarCy2' not found in 'VarCy2'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.C74: Property 'BermTargetYLoc3' not found in 'BermTargetYLoc3'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.B74: Property 'BermTargetXLoc3' not found in 'BermTargetXLoc3'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.B73: Property 'BermTargetXLoc2' not found in 'BermTargetXLoc2'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.B72: Property 'BermTargetXLoc1' not found in 'BermTargetXLoc1'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.U75: Property 'VarPy1' not found in 'VarPy1'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.U74: Property 'VarPy1' not found in 'VarPy1'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.U73: Property 'VarPy1' not found in 'VarPy1'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.U72: Property 'VarPy1' not found in 'VarPy1'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.D109: Property 'NozMntBar3XRot' not found in 'NozMntBar3XRot'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.D107: Property 'Ext2' not found in 'Ext2'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.H106: Property 'NozMntBar1Wth' not found in 'NozMntBar1Wth'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.D105: Property 'Ext1' not found in 'Ext1'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.B63: Property 'SenMntPos' not found in 'SenMntPos'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.D72: Property 'NozMntPltThk' not found in 'NozMntPltThk'
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.L75: '<=' not supported between instances of 'str' and 'Base.Quantity'
in expression: VarCx4 <= VarPx5
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.K75: '<=' not supported between instances of 'str' and 'Base.Quantity'
in expression: VarCx4 <= VarPx4
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.G75: '<' not supported between instances of 'str' and 'Base.Quantity'
in expression: VarCx4 < VarPx5
13:38:17  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.J75: Cannot convert str to Quantity
in expression: VarCD4 / VarCP4_2
[... cut ...]
13:38:24  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.AC72: unsupported operand type(s) for /: 'str' and 'int'
in expression: SprayAngCalc1 / 2
13:38:24  <Spreadsheet> Cell.cpp(642): SpreadsheetIssue___Copy#Spreadsheet.AB72: unsupported operand type(s) for /: 'str' and 'int'
in expression: SprayAngCalc1 / 2
13:38:24  Recompute failed! Please check report view.`

2) When I hit the Refresh button, it DOES NOT resolve the issue.  The errors are still there.  I have to save and close the file, then open it back up, then hit Refresh, and then it fixes it.

chennes, 2022-11-04:
The steps to reproduce #7511 also indicate that Refresh does not resolve the issue, I believe. So I don't think at this time it's necessary to open a whole new bug report for the Undo issue, I'm going to just make sure there's a back-link to this one. Since the basis of this particular ticket was solved I'm going to close it, with the understanding that there is still an undo issue with Spreadsheet.

closed 2022-11-04 by chennes