FreeCAD/FreeCAD #7580: [Bug] Canceling Attachment dialog reverts renaming

wwmayer, 2022-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=630729#p630729

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 18.04.6 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.30591 (Git)
Build type: Debug
Branch: master
Hash: 16f688e933eb849222219eac0a27058508b63035
Python 3.6.9, Qt 5.9.5, Coin 4.0.0a, Vtk 7.1.1, OCC 7.3.0
Locale: German/Germany (de_DE)
[/code]

v0.20 is also affected
```

### Issue description

* New document -> Part WB -> Create a Cube
* Select Cube in TreeView
* Change Label in property Editor (let's say "AAA") -> Tab
* Select Map Mode, click the "..." in the value field, cancel Task dialog => OK, Label is "AAA"
* Change Label in property Editor (let's say "BBB") -> Tab
* Select Map Mode, click the "..." in the value field, cancel Task dialog =>  NOK, Label is "AAA"

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Zolko-123, 2022-10-28:
I can confirm the bug.

@luzpaz : why did you set the priority to "High" ? This seems annoying but quite minimal. High priority should be kept for really bad bugs. This is probably some  forgotten push on the action stack

luzpaz, 2022-10-30:
@Zolko-123 because this is  workflow issue that should be cleaned up before the next release. I distinguished it so it won't get buried

Be-Cool-Man, 2024-07-24:
This behavior is still existent in 0.22dev
```
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git) AppImage
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE) [ OS: German/Germany (de_DE) ]
Installed mods: 
  * BillOfMaterials 0.1.3.0
  * Assembly4 0.50.15
  * A2plus 0.4.65
  * ExplodedAssembly
  * Assembly3 0.12.2
  * sheetmetal 0.4.23
  * fcgear 1.0.0
  * OpenDark 2024.3.13
  * fasteners 0.5.24
  * Help 1.0.3
```

luzpaz, 2024-07-24:
>  In v0.18 all this worked correctly.

Per wwmayer in the forum

YamiGhor, 2024-08-27:
Reproduced the isue:
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Spain (es_ES)
Installed mods: 
  * boltsfc 2022.11.5
  * fasteners 0.5.27
[/code]

zbynekwinkler, 2024-10-04:
I can reproduce the issue as well on 596fa2856b. If I instead use right click on the object, select `Rename`, enter `BBB`, press `enter` - the label survives the trip to the task + cancel.

Can anyone point me to the code handling this? I am still finding my way around. Thanks.

luzpaz, 2024-10-04:
@PaddleStroke can you sleuth out where the code for this functionality is for @zbynekwinkler ?

PaddleStroke, 2024-10-04:
That's it : 

```
void TaskDlgAttacher::open()
{
    Gui::Document* document = Gui::Application::Instance->getDocument(ViewProvider->getObject()->getDocument());
    if (!document->hasPendingCommand())
        document->openCommand(QT_TRANSLATE_NOOP("Command", "Edit attachment"));
}

bool TaskDlgAttacher::reject()
{
    Gui::DocumentT doc(getDocumentName());
    Gui::Document* document = doc.getDocument();
    if (document) {
        // roll back the done things
        document->abortCommand();
        Gui::Command::doCommand(Gui::Command::Gui,"%s.resetEdit()", doc.getGuiDocumentPython().c_str());
        Gui::Command::doCommand(Gui::Command::Doc,"%s.recompute()", doc.getAppDocumentPython().c_str());
    }

    return true;
}
```
The command is opened only if !document->hasPendingCommand(). While it abort it in all cases. So if no command was opened, it will revert the previous command

PaddleStroke, 2024-10-04:
@wwmayer it seems that you introduced this : 
https://github.com/FreeCAD/FreeCAD/commit/3924d410730ecb7c5dcccef87cc815b9b714c105

zbynekwinkler, 2024-10-04:
Hmm, not being fast enough :smile: . I have added `--log-file` argument when starting freecad and I am watching the output with `tail -f` while trying different sequences of steps to better understand the problem.

When I click on `...` next to `Map Mode` I get

```
Log: Active view is Unnamed : 1[*] (at 0x7fffe80050f0)
Log: 281.592 MainWindow.cpp(1704): update actions
Log: <App> AutoTransaction.cpp(139): set active transaction 'Edit attachment'
Log: 292.88 MainWindow.cpp(1704): update actions
```

After clicking `Cancel` I don't see anything similar to "abort transaction" in the log and further editing any other property leaves me with

```
Log: <PropertyView> PropertyEditor.cpp(308): editor already transacting Edit attachment
```

So when I go for `...` next to `Map Mode` again and click `Cancel` again, everything I have changed is rolled back. It seems canceling the task only does rollback but does not really end the transaction.

With that said, I don't understand how #17013 fixes the problem. But I'll rebuild and test and report back.

PaddleStroke, 2024-10-05:
Ah sorry 😄 
It's just that it was an easy fix once I found where it was in the code. If you're looking for small projects though I can find you some others nice ones !

zbynekwinkler, 2024-10-05:
Don't worry. In the meantime, I've tested your fix and it _does_ work even thought I don't understand why (yet). :+1:

wwmayer, 2024-10-06:
> Was there a reason you did not condition the commit and abort ? Perhaps I'm missing something

Undo that change, open and cancel the attachment editor and you will see that it keeps a transaction in the undo list about editing the attachment

commit 12c6fc29 (2024-10-08): Core: Fix #7580 (#17013)

Fixes several undo/redo issues with the attachment dialog
commit 259d21c5 (2024-10-08): Core: Fix #7580 (#17013)

Fixes several undo/redo issues with the attachment dialog
closed 2024-10-08 by wwmayer