FreeCAD/FreeCAD #7582: [Bug] Spreadsheet settings in user.cfg do not work

MisterMakerNL, 2022-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?f=34&t=72294&sid=f35cc4534586974cb1e48ffbe7fd4ece&start=10

### Version

0.21 (Development)

### Full version info

```shell
So after trying to adjust the colors in Spreadsheet by qss theme file and failed I tried do do it try the preference pack cfg file and I also failed.
I tried to modify the settings in the user.cfg file, but that didn't do anything also. So I have to conclude it is broken. I tested it on 0.21 and 0.20.1 on Windows 10.
Below code that was in the user.cfg I modified.

          <FCParamGroup Name="Spreadsheet">
            <FCInt Name="aliasBgColor" Value="746455039"/>
            <FCUInt Name="textFgColor" Value="746455039"/>
            <FCUInt Name="positiveFgColor" Value="2105376255"/>
            <FCUInt Name="negativeFgColor" Value="4294967295"/>
            <FCText Name="ImportExportDelimiter">tab</FCText>
            <FCText Name="ImportExportQuoteCharacter">I</FCText>
            <FCText Name="ImportExportEscapeCharacter">F</FCText>
            <FCUInt Name="aliasBgColor" Value="2105376255"/>
            <FCUInt Name="AliasedCellBackgroundColor" Value="11141375"/>
          </FCParamGroup>

Also relevant:
https://github.com/FreeCAD/FreeCAD/issues/6805#issuecomment-1272336312
```

### Issue description

Trying to modify the alias background color and the text color of spreadsheets, but it seems to be broken.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-10-15:
@chennes can I assign this to you?

chennes, 2022-10-15:
The bug is wrong: the data type of the alias color is FCText, not FCUint.

luzpaz, 2022-10-15:
@MisterMakerNL :point_up:

MisterMakerNL, 2022-10-16:
@luzpaz Yeah, the bug is wrong but kinda doesn't matter since it is still bugged.
The current work around is adding this line to your preference pack:
`            <FCText Name="AliasedCellBackgroundColor">#850097</FCText>`
The issue is that it doesn't get reset anywhere so if you switch back to no stylesheet or no preference pack it stays that color.
Workaround for that currently is apply my "ExtremeProLight" since that resets it too the yellow color.

The bug is that the settings in the user.cfg are not FCtext and use the wrong parameter. Also adding a tab for colors for spreadsheet would help a lot for this, just like other addons have a color tab.

chennes, 2022-10-16:
Adding access to those variables to the preferences dialog is relatively simple (though only relatively, because we can't use the built-in color selection widget, which would save the value as a uint!). Switching to using QSS for the styling is a bit more work.

MisterMakerNL, 2022-10-17:
My suggestion would be to make it similar to the rest. So not using text parameters, although using hex is nice have the same workflow for all settings is nice.

Intuos5, 2023-02-05:
Would be nice to have a colour tab for spreadsheet preferences indeed. Stumbled upon the Aliased cell color issue in my theming efforts as well.

luzpaz, 2023-12-17:
What's the roadmap here ?

maxwxyz, 2024-07-14:
Is this still relevant?

obelisk79, 2024-08-17:
I'm finding this issue difficult to understand. What is the actual nature of the problem? that colors don't change when switching between themes? Assuming this would be experienced when sharing files.

maxwxyz, 2024-08-18:
@MisterMakerNL could you weigh in?

Intuos5, 2024-08-18:
IIRC I could not change the spreadsheet colours using the old stylesheets, but that may be on me.

MisterMakerNL, 2024-08-18:
This is fixed, somewhere between then an now, maybe by @Syres916 reset classic pr.
Also I now could easily fix this myself :) I learned a lot in pasty 1,5 year!

closed 2024-08-18 by MisterMakerNL