FreeCAD/FreeCAD #7584: [Bug] Drawing large STEP files is slow

flaviut, 2022-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

None, but I read the code :)

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: makepkg
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.7, Qt 5.15.6, Coin 4.0.1, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Part

### Issue description

When I [open the attached file](https://github.com/FreeCAD/FreeCAD/files/9742272/ender-5.zip)[^1], FreeCAD is slow to render as I rotate the file around.

Digging in a bit, it seems like each polygon is being drawn one-by-one, rather than with a VBO. Making all these draw calls is very slow. It looks like [there's VBO support in this code, but it is unconditionally disabled](https://github.com/FreeCAD/FreeCAD/blob/42a19ab883e4818f57d2c0cd48affbbb4e734500/src/Mod/Part/Gui/SoBrepFaceSet.cpp#L650).

There's also quite a bit of other stuff happening due to the mouse movement events. You can see it in the profile trace, but I'm less confident about what's going on there.

[^1]: source is https://grabcad.com/library/creality-ender-5-pro-1

### Anything else?

[Profiler trace](https://github.com/FreeCAD/FreeCAD/files/9742278/profile.zip) (extract and open with [speedscope.app](https://www.speedscope.app/))

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2022-10-10:
This shouldn't be marked as a bug, more like a performance enhancement request. Please check if you have VBO option is enabled in `Preference -> Display -> Use OpenGL VBO`. The code you linked is only active (i.e. `!ctx2`) when you use `SubLink` to partial rendering some larger shape, which is not what happened here. 

On my computer, the model you linked renders about 15 fps without VBO, and 25fps with VBO. There are tons of rendering enhancement done in my [branch](https://github.com/realthunder/FreeCAD), and they will land in upstream, eventually. Just give it some time. The model renders about 75 fps in my branch. It can go higher because right now we cap the redraw rate with monitor refresh rate, although it is unnecessary as we are not playing games here.

flaviut, 2022-10-10:
> Please check if you have VBO option is enabled in Preference -> Display -> Use OpenGL VBO

You're right, I had that disabled. Enabling it makes things much faster.

I would have liked to talk about enabling it by default, [but it looks like that discussion has already happened](https://forum.freecadweb.org/viewtopic.php?style=3&f=34&t=70838) and it's simply a matter of lots of work to be done and not enough time :)

> On my computer, the model you linked renders about 15 fps without VBO, and 25fps with VBO.

Interesting--now that I open the file again, I get a similar experience. Before I was seeing 3fps or so. I'm not sure what changed. And your branch is buttery smooth, thank you for working on that!

> The code you linked is only active (i.e. !ctx2) when you use SubLink to partial rendering some larger shape, which is not what happened here.

All I know is that it was active when I was doing my testing earlier:

[image]

But I did misread the code, it is assigned to by reading the enabled GL_EXTENSIONS: https://github.com/FreeCAD/FreeCAD/blob/42a19ab883e4818f57d2c0cd48affbbb4e734500/src/Mod/Part/Gui/SoBrepFaceSet.cpp#L516-L517

Maybe a driver bug or a race condition or something in my earlier run?

But on the other hand, it doesn't seem like this follows the VBO setting in the preferences.

luzpaz, 2022-10-15:
@realthunder how should this be re-phrased as a feature request then ?

realthunder, 2022-10-15:
3D Rendering Performance Enhancement, perhaps? This is not limited to STEP files, but any big models.

Zolko-123, 2022-10-27:
This Issue can be resumed to "make FreeCAD faster". Also, the user has obviously no clue about the real reason (didn't even switch VBO). This is no specific request and should be closed to clean up the Issue list

luzpaz, 2022-11-03:
Closing as per zolko's feedback

closed 2022-11-03 by luzpaz